Arksine/moonraker #3: API Endpoints for RESTful

puneit-hassio, 2020-08-03:
Hello, 
I am trying to create a RESTful sensor in Home Assistant to get the progress of the print and then eventually get notified when print finishes. I am unable to find the method to do so from the documentation.
It will be great to have the following information available from REST
1. Status of printer - (standby, printing, error, waiting for user input...)
2. Percentage of a print job from 0 to 100
3. Print job time remaining 
Thank you

Arksine, 2020-08-04:
Its possible to get this information through a combination of the `/printer/info`  and `/printer/objects/status` APIs.  The state you can fetch from `/printer/objects/status` isn't yet documented but its own my todo list.

Currently I am working on merging the support functionality necessary to communicate with the official Klipper repo so it isn't necessary to pull the `dev-moonraker-testing` branch of my fork.  Included in this will be a more streamlined approach to fetching the state of a print.  When this is merged I will update this issue with an example of how to fetch the data you are looking for.

Arksine, 2020-10-28:
It's been a while since this issue was created, but I thought I would update it.  The API is relatively stable now and initial API documentation is complete.   An outline of how to accomplish your request is [available here](https://github.com/Arksine/moonraker/blob/master/docs/web_api.md#basic-print-status).   The entire API is available in web_api.md, and printer_objects.md contains information about "klipper printer objects" that can be queried from the API.

Arksine, 2021-01-03:
As the discussion here seems to have concluded I'm going to close this.  As mentioned above, the API documentation contains an example of how to retrieve status from moonraker.

closed 2021-01-03 by Arksine