Arksine/moonraker #7: API not working.

changedsoul, 2020-08-12:
Sorry the title is possibly so vague. During installing, I get to the step to open url: "http://<printer-ip>:7125/printer/info"
If I see this: "{"result": {"hostname": "voron250", "error_detected": false, "version": "v0.8.0-643-g528f9f25", "is_ready": true, "message": "Printer is ready", "cpu": "4 core ARMv7 Processor rev 4 (v7l)"}}"
My API is working....I do not get this, so I guess my API is not working.

I have this in my moonraker log file.:
cat /tmp/moonraker.log 
Moonraker Log Start...
Git Version: d04a419
================================================================================
2020-08-12 22:00:17,803 [moonraker.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 393, in main
    server = Server(cmd_line_args)
  File "/home/pi/moonraker/moonraker/moonraker.py", line 34, in __init__
    config = confighelper.get_configuration(self, args)
  File "/home/pi/moonraker/moonraker/confighelper.py", line 71, in get_configuration
    raise ConfigError(f"Configuration File Not Found: '{cfg_file_path}''")
confighelper.ConfigError: Configuration File Not Found: '/home/pi/moonraker.conf''

Any help would be appreciated.
I am running a fresh install of Rasperian Debian lite version updated to most recent packages, and have pulled what ever gets pulled when following these instructions: https://github.com/meteyou/mainsail/blob/master/docs/installation.md

Arksine, 2020-08-12:
Moonraker recently received some updates that allow it to function with the official Klipper repo.  Unfortunately these updates broke compatibility with Mainsail, as mentioned in [moonraker's documentation](https://github.com/Arksine/moonraker/blob/master/docs/installation.md#installation).

You can however check out an compatible version of Moonraker to get you up and running until Mainsail is updated:
```
cd ~/moonraker
git fetch --tags
git checkout tags/v.08
sudo service moonraker restart
sudo service klipper restart
```

changedsoul, 2020-08-12:
Thanks, that got me closer. Now i can continue on with installing the rest.

closed 2020-08-18 by changedsoul