Arksine/moonraker #9: Can't install successfully

jbhewitt, 2020-08-18:
Not sure what the next step is to diagnose what's going wrong here.

I had this upgrading Mainsail to the latest version, so I've tried with a fresh copy of raspberyos and same problem.

I follow the installation instructions and no errors. 

When I search the klippy.log file i do NOT see a "New connection" - clearly moonraker isn't functioning.

browser url  - http://192.168.86.51:7125/printer/restart

moonraker.log
`2020-08-18 11:58:51,084 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.86.47
2020-08-18 11:58:51,088 [web.py:log_request()] - 405 GET /printer/restart (192.168.86.47) 6.69ms`

Browser error
`Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1703, in _execute
    result = await result
  File "/home/pi/moonraker/moonraker/app.py", line 282, in get
    raise tornado.web.HTTPError(405)
tornado.web.HTTPError: HTTP 405: Method Not Allowed`

jbhewitt, 2020-08-18:
here is the install script info - I can't see any issues with it

`
pi@raspberrypi:~/moonraker/scripts $ ./install-moonraker.sh 
Klipper service found!

###### Running apt-get update...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done                 

###### Installing packages...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
nginx is already the newest version (1.14.2-2+deb10u2).
python3-dev is already the newest version (3.7.3-1).
python3-virtualenv is already the newest version (15.1.0+ds-2).
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

###### Updating python virtual environment...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: tornado==6.0.4 in /home/pi/moonraker-env/lib/python3.7/site-packages (from -r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 2)) (6.0.4)
Requirement already satisfied: pyserial==3.4 in /home/pi/moonraker-env/lib/python3.7/site-packages (from -r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 3)) (3.4)

###### Installing system start script...

###### Launching Moonraker API Server...
[ ok ] Stopping klipper (via systemctl): klipper.service.
[ ok ] Restarting moonraker (via systemctl): moonraker.service.
[ ok ] Starting klipper (via systemctl): klipper.service.
`

Arksine, 2020-08-18:
A recent change to Klipper broke compatibility with moonraker/mainsail.  We are working on updating our end to fix things.   Kevin has more changes in the works that we need to accommodate as well.  When the API is baked on Klipper's side I can make an official release.  For the time being behavior may be be a bit unpredictable.

jbhewitt, 2020-08-18:
Ahh ok thanks for the update - much appreciated!

Arksine, 2020-09-11:
This should now be resolved, as Moonraker works with the official Klipper repo and the latest version of Mainsail.

closed 2020-09-11 by Arksine