Arksine/moonraker #10: Keep getting 404 with printer/info

LeandroMarceddu, 2020-08-30:
As the title says. Logs include only these:
21:44:55,290 [moonraker.py:_read_klippy_stream()] - Unknown command received: {"id": 3048972656, "error": {"message": "webhooks: No registered callback for path 'GET'", "error": "WebRequestError"}}

klippylog has only these:
webhooks: No registered callback for path 'GET'

Arksine, 2020-08-30:
Make sure you have pulled the latest updates for both Klipper and Moonraker.   There were some significant changes to Klipper's "webhooks" functionality that required changes to Moonraker.  The master branch of both Klipper and Moonraker should now be in sync.  I will tag an official beta release soon.

Be aware that you do need to make some changes to Klipper's "default" file, as noted in the installation instructions.  It is unlikely that installers have included this since these changes were just merged yesterday.  See the [installation documentation](https://github.com/Arksine/moonraker/blob/master/docs/installation.md) for details on this.

LeandroMarceddu, 2020-08-30:
Yup, seems you're right. I didn't know what but deleting everything and starting fresh actually did it. Thanks, my bad!

closed 2020-08-30 by LeandroMarceddu