Arksine/moonraker #11: Connecting a PanelDue to Pi3

paulhew69, 2020-09-03:
Hi,
Thanks for your time.
I have dug around trying to help myself and learn!
My PanelDue is running firmware 1.24

So the pins I need to use in the Pi are 4, 6, 8, 10
4 is +5v, 6 is Gnd. 
8 is UART0 Tx, 10 is UART0 Rx.

Ran sudo raspi-config and disabled console messages and enabled serial.

ls -l /dev shows me
`crw-rw---- 1 root dialout 166,   0 Sep  3 17:17 ttyACM0
crw-rw---- 1 root dialout 204,  64 Sep  3 17:08 ttyAMA0
`
I modified the moonraker.conf and set serial to 
`[paneldue]
serial: /dev/ttyAMA0` I think this is correct.

Checked moonraker logs file
I have these entries
`2020-09-03 17:22:27,481 [paneldue.py:_connect()] - Attempting to connect to: /dev/ttyAMA0
2020-09-03 17:22:27,483 [paneldue.py:_connect()] - PanelDue Connected
2020-09-03 17:22:27,485 [moonraker.py:connect()] - Klippy Connection Established`

I have rebooted numerous times and the PanelDue just says 'Connecting' in the top right corner.

Have I missed something in my process?

Your input would be appreciated.

Regards,

Paul

paulhew69, 2020-09-03:
I tried a little harder and read some more documents.
Pi3 serial port is 'ttyS0'

Homing works, I can heat up the hotend, list files.

Superb and thank you for your hard work.

closed 2020-09-03 by paulhew69

robthide37, 2020-11-02:
Where did you find info on how to set this up?

paulhew69, 2020-11-02:
In the PanelDue addon and some great help from Arksine>
There is another post by me where I was struggling and losing connection.
Arksine guided me to use this.
https://www.amazon.co.uk/gp/product/B07K76Q2DX/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1

HTH

Paul.

robthide37, 2020-11-02:
you had to use 2 of them ? or you were able to use the raspberry pi usb port then run it to the paneldue off those pins since it would be 5v logiv i guess and be able to run longer distance vs the 3.3 not really cutting it ?

robthide37, 2020-11-03:
> In the PanelDue addon and some great help from Arksine>
> There is another post by me where I was struggling and losing connection.
> Arksine guided me to use this.
> https://www.amazon.co.uk/gp/product/B07K76Q2DX/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1
> 
> HTH
> 
> Paul.

What does the address become when using this adapter?

Arksine, 2020-11-03:
@robthide37 The configuration info is available [here](https://github.com/Arksine/moonraker/blob/master/docs/installation.md#paneldue-plugin).

IIRC Paul had to use the USB adapter because his UART run was too long (or perhaps too noisy).  Typically those devices enumerate to `/dev/ttyUSBx` where x is decided based on the number of USB ttys on your system (so starts at `/dev/ttyUSB0`, etc).

You can use the Pi GPIOs as long as the run isn't too long.  You would want the wires to avoid noise if possible.  I use the GPIOs, however I use the Pi's PL011 instead of the miniuart, which is on `/dev/ttyAMA0`.  You typically need to go through a few steps to get the PL011 set up as the primary UART:
- Disable the serial console in `raspi-config`
- run `sudo systemctl disable hciuart`
- edit /boot/config.txt  and add:
  ```
  dtoverlay=disable-bt
  enable_uart=1
  ```
- reboot pi

The [Pi documentation](https://www.raspberrypi.org/documentation/configuration/uart.md) has more details on it.

One thing to remember, the PanelDue needs +5v power, but the Rx and Tx pins use 3.3v logic.  Not a problem when connecting to the Pi, but its something to be aware of when using a USB-TTL adapter.

robthide37, 2020-11-04:
Thankyou very much. I know the mini uart has worse buffers well smaller and
nicer hardware flow control so its better to use the other one . Only
downside is losing Bluetooth right ? I am going to do exactly what
you suggested btw thankyou very much I will be pledging or do you have al
patreon ?