Arksine/moonraker #15: Malformed 'START_PRINT'

taw-pi, 2020-09-06:
Not sure if this is a bug or not but when issuing G-codes from a file there is, in some instances, a semi-colon at the end of the G-code to start a comment. I had START_PRINT; in my G-code file referring to my klipper printer.cfg macro START_PRINT. 
Using Octoprint the semicolon did not cause any problems, however I got the message Malformed 'START_PRINT' using
with moon raker/mainsail. Removing the semicolon from the G-code file solved the issue.

Arksine, 2020-09-06:
Moonraker does not parse gcode files directly in the way Octoprint does.  Klipper's "virtual_sdcard" functionality is used for gcode file processing.  It is possible that Octoprint removes that semicolon prior to sending the command to Klipper, however having a Semicolon directly after a gcode is technically invalid, as the semicolon is read as part of the command and not a comment.

taw-pi, 2020-09-06:
Thanks for the clarification, I will close this issue now.

closed 2020-09-06 by taw-pi