Arksine/moonraker #16: Can not access virtual sd card

zellneralex, 2020-09-09:
Hi I am running out of ideas. I can not access the virtual SD Card. I already reinstalled everything twice.

Printer.cfg
[virtual_sdcard]
path: ~/sdcard

klippy.log
[virtual_sdcard]
path = ~/sdcard

moonraker.log

2020-09-09 07:26:50,594 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "result": {"status": {"temperature_sensor bed": {"measured_min_temp": -94.543263861882, "measured_max_temp": 0.0, "temperature": -94.543263861882}, "heaters": {"available_sensors": ["heater_bed", "temperature_sensor bed", "temperature_fan chamber", "extruder"], "available_heaters": ["heater_bed", "extruder"]}, "fan": {"speed": 0.0}, "bed_mesh": {"mesh_max": [0.0, 0.0], "mesh_matrix": [[]], "profile_name": "", "mesh_min": [0.0, 0.0], "probed_matrix": [[]]}, "pause_resume": {"is_paused": false}, "display_status": {"progress": 0.0, "message": null}, "gcode_move": {"homing_origin": [0.0, 0.0, 0.0, 0.0], "speed_factor": 1.0, "gcode_position": [0.0, 0.0, 0.0, 0.0], "absolute_extrude": true, "absolute_coordinates": true, "position": [0.0, 0.0, 0.0, 0.0], "speed": 1500.0, "extrude_factor": 1.0}, "print_stats": {"print_duration": 0.0, "total_duration": 0.0, "filament_used": 0.0, "filename": "", "state": "standby", "message": ""}, "heater_fan controller_fan": {"speed": 0.0}, "temperature_fan chamber": {"speed": 0.0, "temperature": -94.543263861882, "target": 50.0}, "idle_timeout": {"state": "Idle", "printing_time": 0.0}, "query_endstops": {"last_query": {}}, "toolhead": {"square_corner_velocity": 5.0, "max_accel": 3000.0, "homed_axes": "", "estimated_print_time": 612.4568908583333, "max_velocity": 300.0, "print_time": 0.001, "max_accel_to_decel": 1500.0, "position": [0.0, 0.0, 0.0, 0.0], "extruder": "extruder"}, "configfile": {"config": {"tmc2209 extruder": {"sense_resistor": "0.110", "interpolate": "false", "stealthchop_threshold": "0", "run_current": "0.5", "microsteps": "16", "uart_pin": "P1.4", "hold_current": "0.4"}, "display_data __voron_display printing_time": {"position": "2, 10", "text": "\n{% set ptime = printer.idle_timeout.printing_time %}\n{ \"%02d:%02d\" % (ptime // (60 * 60), (ptime // 60) % 60) }"}, "display_data __voron_display progress_text2": {"position": "1, 10", "text": "\n{% set progress = printer.display_status.progress %}\n{ draw_progress_bar(1, 10, 6, progress) }"}, "probe": {"samples_tolerance_retries": "3", "x_offset": "0", "pin": "^z:P0.10", "sample_retract_dist": "3.0", "samples_tolerance": "0.006", "samples_result": "median", "y_offset": "25.0", "samples": "3", "z_offset": "0", "speed": "10.0"}, "stepper_z2": {"step_pin": "z:P0.22", "step_distance": "0.00125", "dir_pin": "!z:P2.11", "enable_pin": "!z:P0.21"}, "neopixel neo_display": {"pin": "z:P1.21", "chain_count": "3", "color_order_grb": "False", "initial_red": "0.1", "initial_green": "0.5", "initial_blue": "0.0"}, "gcode_macro UNLOAD_FILAMENT": {"gcode": "\n{% if printer.toolhead.status == \"Ready\" %}\nM83                            ; set extruder to relative\nG1 E10 F300                    ; extrude a little to soften tip\nG1 E-780 F1800                 ; retract filament completely\nM82                            ; set extruder to absolute\n{% else %}\n{ printer.gcode.action_respond_info(\"Filament unloading disabled while printing!\") }\n{% endif %}"}, "mcu rpi": {"serial": "/tmp/klipper_host_mcu"}, "display_data __voron_display chamber": {"position": "2, 0", "text": "\n{% set chamber = printer['temperature_fan chamber'] %}\n~chamber~\n{ \"%3.0f\" % (chamber.temperature,) }\n~degrees~"}, "heater_bed": {"control": "pid", "pid_kp": "58.437", "sensor_pin": "z:P0.25", "pid_kd": "363.769", "heater_pin": "z:P2.3", "sensor_type": "NTC 100K MGB18-104F39050L32", "max_power": "0.6", "pid_ki": "2.347", "min_temp": "0", "max_temp": "120"}, "pause_resume": {}, "virtual_sdcard": {"path": "~/sdcard"}, "stepper_z1": {"step_pin": "z:P0.19", "step_distance": "0.00125", "dir_pin": "z:P0.20", "enable_pin": "!z:P2.8"}, "gcode_macro LOAD_FILAMENT": {"gcode": "\n{% if printer.toolhead.status == \"Ready\" %}\nM83                            ; set extruder to relative\nG1 E750 F1800                  ; quickly load filament to down bowden\nG1 E30 F300                    ; slower extrusion for hotend path\nG1 E15 F150                    ; prime nozzle with filament\nM82                            ; set extruder to absolute\n{% else %}\n{ printer.gcode.action_respond_info(\"Filament loading disabled while printing!\") }\n{% endif %}"}, "verify_heater extruder": {"max_error": "120", "check_gain_time": "20", "heating_gain": "2", "hysteresis": "5"}, "display_data __voron_display bed": {"position": "1, 0", "text": "{ render(\"_vheater_temperature\", param_heater_name=\"heater_bed\") }"}, "stepper_z": {"homing_retract_dist": "3.0", "position_endstop": "-0.5", "position_max": "340", "endstop_pin": "z:P1.27", "second_homing_speed": "3.0", "step_pin": "z:P2.2", "position_min": "-5", "homing_speed": "15.0", "dir_pin": "!z:P2.6", "step_distance": "0.00125", "enable_pin": "!z:P2.1"}, "stepper_y": {"homing_positive_dir": "true", "homing_retract_dist": "5", "position_endstop": "350", "position_max": "350", "endstop_pin": "^!P1.28", "step_pin": "P0.19", "position_min": "0", "homing_speed": "25", "dir_pin": "!P0.20", "step_distance": "0.00625", "enable_pin": "!P2.8"}, "stepper_x": {"homing_positive_dir": "true", "homing_retract_dist": "5", "position_endstop": "350", "position_max": "350", "endstop_pin": "^!P1.29", "step_pin": "P2.2", "position_min": "0", "homing_speed": "25", "dir_pin": "!P2.6", "step_distance": "0.00625", "enable_pin": "!P2.1"}, "tmc2209 stepper_x": {"sense_resistor": "0.110", "interpolate": "True", "stealthchop_threshold": "0", "run_current": "0.8", "microsteps": "16", "uart_pin": "P1.10", "hold_current": "0.7"}, "display_data __voron_display print_status": {"position": "3, 0", "text": "\n{% if printer.display_status.message %}\n{ printer.display_status.message }\n{% elif printer.idle_timeout.printing_time %}\n{% set pos = printer.toolhead.position %}\n{ \"X%-4.0fY%-4.0fZ%-5.2f\" % (pos.x, pos.y, pos.z) }\n{% else %}\n{ \"VX.xxx \" }\n~voron~\n{% endif %}"}, "heater_fan controller_fan": {"heater": "heater_bed", "kick_start_time": "0.5", "pin": "z:P2.4", "heater_temp": "45.0"}, "display_data __voron_display progress_text": {"position": "1, 10", "text": "\n{% set progress = printer.display_status.progress %}\n{ \"{:^6.0%}\".format(progress) }"}, "display_glyph chamber": {"data": "\n0000000000000000\n1111111111111111\n1000010000100001\n1000010000100001\n1000011111100001\n1000000000000001\n1000000000000001\n1000001111000001\n1011101001011101\n1000001111000001\n1000000110000001\n1000000000000001\n1011111111111101\n1000100000010001\n1111111111111111\n0000000000000000"}, "display_glyph voroninv": {"data": "\n0000001110000000\n0000111111100000\n0001111111110000\n0111111111111100\n1111100111001110\n1111001110011110\n1110011100111110\n1100111001111110\n1111110011100110\n1111100111001110\n1111001110011110\n1110011100111110\n0111111111111100\n0001111111110000\n0000111111100000\n0000001110000000"}, "tmc2209 stepper_z1": {"sense_resistor": "0.110", "interpolate": "true", "stealthchop_threshold": "0", "run_current": "0.8", "microsteps": "16", "uart_pin": "z:P1.9", "hold_current": "0.8"}, "tmc2209 stepper_z3": {"sense_resistor": "0.110", "interpolate": "true", "stealthchop_threshold": "0", "run_current": "0.8", "microsteps": "16", "uart_pin": "z:P1.4", "hold_current": "0.8"}, "tmc2209 stepper_z2": {"sense_resistor": "0.110", "interpolate": "true", "stealthchop_threshold": "0", "run_current": "0.8", "microsteps": "16", "uart_pin": "z:P1.8", "hold_current": "0.8"}, "temperature_sensor bed": {"min_temp": "0", "sensor_type": "ATC Semitec 104GT-2", "gcode_id": "1B", "sensor_pin": "z:P0.23", "max_temp": "120"}, "gcode_macro PRINT_END": {"gcode": "\nM400                           ; wait for buffer to clear\nG92 E0                         ; zero the extruder\nG1 E-4.0 F3600                 ; retract filament\nG91                            ; relative positioning\nG0 Z1.00 X20.0 Y20.0 F20000    ; move nozzle to remove stringing\nTURN_OFF_HEATERS\nM107                           ; turn off fan\nG1 Z5 F3000                    ; move nozzle up 5mm\nG90                            ; absolute positioning\nG0  X175 Y350 F3600            ; park nozzle at rear\nM117 Finished!                 ; display message\nBED_MESH_CLEAR"}, "printer": {"square_corner_velocity": "5.0", "max_accel": "3000", "max_z_accel": "350", "max_v [... cut ...]
2020-09-09 07:26:56,298 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 400, "message": "Invalid base path (gcodes)"}, "id": 6376}
2020-09-09 07:27:07,379 [web.py:log_exception()] - 400 POST /server/files/upload (192.168.178.32): Gcodes root not available
2020-09-09 07:27:07,392 [web.py:log_request()] - 400 POST /server/files/upload (192.168.178.32) 14.94ms

The folder ~/sdcard is there and accesable with the user pi

[klippy.log](https://github.com/Arksine/moonraker/files/5192829/klippy.log)
[moonraker.log](https://github.com/Arksine/moonraker/files/5192830/moonraker.log)

I have no idea where to look next

zellneralex, 2020-09-09:
output of http://voron350.local/server/files/list?root=gcodes

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/plugins/file_manager.py", line 385, in get_file_list
    filelist = self._update_file_list(base)
  File "/home/pi/moonraker/moonraker/plugins/file_manager.py", line 263, in _update_file_list
    raise self.server.error(msg)
utils.ServerError: No known path for root: gcodes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/app.py", line 311, in _process_http_request
    result = await self.callback(self.request.path, method, args)
  File "/home/pi/moonraker/moonraker/plugins/file_manager.py", line 104, in _handle_filelist_request
    return self.get_file_list(format_list=True, base=root)
  File "/home/pi/moonraker/moonraker/plugins/file_manager.py", line 389, in get_file_list
    raise self.server.error(msg)
utils.ServerError: Unable to update file list

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1703, in _execute
    result = await result
  File "/home/pi/moonraker/moonraker/app.py", line 290, in get
    await self._process_http_request('GET')
  File "/home/pi/moonraker/moonraker/app.py", line 314, in _process_http_request
    e.status_code, str(e)) from e
tornado.web.HTTPError: HTTP 400: Bad Request (Unable to update file list)

jakep82, 2020-09-09:
I searched the klippy.log file you've posted, and I don't see virtual_sdcard anywhere in it.  Are you sure you edited the printer.cfg in the correct location when you defined that section?

zellneralex, 2020-09-09:
I have it working again! Root Cause if klipper is in error state Moonraker seams not to read the virtual_sdcard location. As soon I did a few tweaks to get klipper (temporarily) to pass, I am able to access the gcode folder and save modifications to the gui. So you need a working klipper. Maybe you could add a note to the install doc

zellneralex, 2020-09-09:
Jakep82 I Ifind the virtual_sdcard 2 times in the log (2 klipper restarts) always near the end of the config section 
[virtual_sdcard]
path = ~/sdcard

[pause_resume]

[display_status]
=======================
Extruder max_extrude_ratio=0.266081

But as already mentioned problem solved klipper must have The Status ready for moonraker to read that location

zellneralex, 2020-09-10:
This can be closed, as soon you modify the manual or decide as it good as is

Arksine, 2020-09-11:
I have advised meteyou to use the config folder for storing gui.json, as it is available immediately if configured.  I will update the docs and close this issue when that is complete.

commit 36fb1da4 (2020-09-12): docs:  add missing documentation

- Document "enable_cors".  Resolves #18.
- Add a note that Moonraker is not aware of the gcode file path  until Klippy successfully starts.  Resolves #16.
closed 2020-09-12 by Arksine