Arksine/moonraker #17: Feature Request Make the GPIOs defined in the Power section controllable via klipper

zellneralex, 2020-09-10:
I start here, I know the FR needs modification at klipper and moonraker

I like the idea to have gpio relays controllable even if klipper is not running but it would be helpful if there would be a way that you also use them in gcode_macros. The best example here would be switch off the 24V power supply in you print_end macro.

what you would need in klipper is a definition e.g
[moonrakker_power]
devices: PSU

And then you could use the SET_PIN gcode to manipulate it. Instead sending it to a physical klipper would send it via API. and moonraker would use the pins defined in the power section. Restriction the device must be equal named. Moonraker should ignore calls for devices not specified in the [power] section.

Arksine, 2020-09-11:
I'm not sure that having Klipper toggle GPIOs via calls to Moonraker is the appropriate action here considering Klipper already has the ability to do this via the linux process.  That said, I do understand the request.  Ideally it would be good if Moonraker and Klipper can share some functionality with regard to toggle GPIOs.  This is something I would need to discuss with Kevin to see if he is interested in this kind of functionality and what the best approach is.

zellneralex, 2020-09-11:
Thank you for the feedback

Arksine, 2020-11-08:
FWIW, it is now possible to toggle devices from a Klipper gcode_macro, [documented here](https://github.com/Arksine/moonraker/blob/master/docs/installation.md#power-control-plugin).

closed 2020-11-18 by zellneralex