Arksine/moonraker #18: Is it possible to allow CORS?

skuridin, 2020-09-11:
Hi! I'm curious if it is possible to allow CORS for WebSockets? I see there is a check in the websockets.py but I don't get where this setting can be set.

Arksine, 2020-09-11:
When CORS is enabled via the `enable_cors` option it is enabled for all HTTP requests, including the request to open the websocket.

skuridin, 2020-09-12:
@Arksine Yeah, I get that. The question is how/where to set this option? It's not documented afaik.

Arksine, 2020-09-12:
It is set in the [server] section of moonraker.conf:
```
[server]
enable_cors: True
```
I had thought it was documented, apparently I left it out.  I'll add it to the installation documentation.

closed 2020-09-12 by Arksine

CapnBry, 2021-01-15:
Because this is the first google result for moonraker CORS, for posterity I will note that this option is now in the authorization section as cors_domains. The documentation has it correctly but just in case someone finds themselves here the same way I did-- by not reading the documentation. The value is a regular expression but also supports just `*` for allow all
```
[authorization]
cors_domains: example.org
```

commit 36fb1da4 (2020-09-12): docs:  add missing documentation

- Document "enable_cors".  Resolves #18.
- Add a note that Moonraker is not aware of the gcode file path  until Klippy successfully starts.  Resolves #16.