Arksine/moonraker #19: PanelDue loses sync.

paulhew69, 2020-09-13:
Printing some bits, and the panel **was** is sync with MainSail.
Just happened to look at it and it has lost sync. Still has a 'Printing' status in top right corner.

Do I need to tweak something?

Regards,
Paul

[image]
[image]

Arksine, 2020-09-13:
Can you provide /tmp/moonraker.log?  If you are connected to the Pi's GPIOs with a long run its possible that the PD lost the connections

paulhew69, 2020-09-13:
Hi Eric,

Thanks for the speedy response.
the cable between the PD and Pi is about 1m.

The only other thing I thought of is I am using a buck convertor for each of the devices. Do they need a common ground?

Regards,

Paul.

Arksine, 2020-09-13:
Yes, they should have a common ground.  A 1M cable should be fine at the default baud rate, just make sure it isn't routed near a source of interference such as a motor or motor cable.  If necessary you can shorten the run by connecting the PanelDue to a USB-TTL converter, like the [FT232R](https://www.amazon.com/HiLetgo-FT232RL-Converter-Adapter-Breakout/dp/B00IJXZQ7C/ref=sr_1_4?dchild=1&keywords=ft232rl&qid=1599995048&sr=8-4).

paulhew69, 2020-09-19:
Thanks for the reply, appreciated.

I bought one of these
https://www.amazon.co.uk/DSD-TECH-SH-U09F-Adapter-Featuring/dp/B07K76Q2DX/ref=sr_1_7?dchild=1&keywords=ft232rl&qid=1600502226&sr=8-7
Did I need one where I can inject power from an external source?

It seems to be working, just need to find out how to set the speed to 57600 when the Pi boots.

Arksine, 2020-09-19:
>Did I need one where I can inject power from an external source?

That largely depends on if your Pi can provide enough current to power the PD.   Just keep in mind that the PD needs to a 5v power source, while the UART relies on 3.3v logic.

> seems to be working, just need to find out how to set the speed to 57600 when the Pi boots.

The speed should be set by Moonraker when it makes the connection.  It defaults to 57600, but can be changed in moonraker.conf.

paulhew69, 2020-09-22:
Update, all working superbly now. Reliable connection.

Thanks for your help, greatly appreciated.
Regards

Paul.

closed 2020-09-22 by paulhew69