Arksine/moonraker #21: Parsing object height from PrusaSlicer/SuperSlicer generated G-Code is incorrect in case of vase mode

th33xitus, 2020-09-24:
There is a little issue with the method how g-code from PrusaSlicer is parsed.
`return self._parse_max_float(r"G1\sZ\d+\.\d*\sF", self.footer_data)`
https://github.com/Arksine/moonraker/blob/master/scripts/extract_metadata.py#L139

The regex there expects a speed value at the end, which is not always present for each z height.
I encountered that issue with a model sliced in vase mode which is 50mm high but in Mainsail only 49.90mm are displayed:
[image]

The last few g-code lines are as follows:
```
G1 Z49.800 F7800.000
; printing object heattower_medium.stl id:0 copy 0
G1 F3600.000
G1 Z49.800 X81.249 Y74.249 E0.00179
G1 Z49.825 X130.750 Y74.249 E0.88636
G1 Z49.850 X130.750 Y123.751 E0.88636
G1 Z49.875 X81.249 Y123.751 E0.88636
G1 Z49.900 X81.249 Y74.349 E0.88457
; stop printing object heattower_medium.stl id:0 copy 0

G1 Z49.900 F7800.000
; printing object heattower_medium.stl id:0 copy 0
G1 F3600.000
G1 Z49.900 X81.249 Y74.249 E0.00179
G1 Z49.925 X130.750 Y74.249 E0.88636
G1 Z49.950 X130.750 Y123.751 E0.88636
G1 Z49.975 X81.249 Y123.751 E0.88636
G1 Z50.000 X81.249 Y74.349 E0.88457
```
You can see that the speed is only set each 0.1mm increments, which is a full layerheight. But when reaching the total object height at Z50.000 there is no speed parameter given. Therefore it can't get parsed correctly.
This seems to be an edge case on vase mode prints?

I sliced the same object with vase mode disabled and i get the correct total object height displayed after that:
[image]

The g-code sequence looks there as follows:
```
G1 Z49.900 F7800.000
; printing object heattower_medium.stl id:0 copy 0
G1 X130.650 Y123.751
G1 F3600.000
G1 X81.249 Y123.751 E0.88457
G1 X81.249 Y74.249 E0.88636
G1 X130.750 Y74.249 E0.88636
G1 X130.750 Y123.751 E0.88636
G1 X130.726 Y123.751 E0.00045
; stop printing object heattower_medium.stl id:0 copy 0

G1 Z50.000 F7800.000
; printing object heattower_medium.stl id:0 copy 0
G1 X130.650 Y123.751
G1 F3600.000
G1 X81.249 Y123.751 E0.88457
G1 X81.249 Y74.249 E0.88636
G1 X130.750 Y74.249 E0.88636
G1 X130.750 Y123.751 E0.88636
G1 X130.726 Y123.751 E0.00045
; stop printing object heattower_medium.stl id:0 copy 0
```
You can see that now Z50.000 has an appended F parameter and can be parsed.

Since i am a total noob in regards to RegEx, i can only report this problem and can't provide a solution for it.

I attached the gcode files if you want to take a look at the g-code yourself.
[gcode.zip](https://github.com/Arksine/moonraker/files/5278210/gcode.zip)

Arksine, 2020-09-26:
Thanks.  Truthfully the current implementation isn't the best, as it can also capture a Z-Hop in the end if it isn't useful.  Unfortunately removing the Feedrate parameter would also be unreliable as it increases the chance of capturing position in the END_GCODE.

What I can do is capture height from the default comments supplied by PrusaSlicer:
```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
```
PrusaSlicer and its forks seem to consistently add this pattern.  Sometimes it does not include the G92, but I can account for that.  If the user edits this I can't extract it, in which case I will fall back on the current method.

th33xitus, 2020-09-27:
Thank you for figuring out a workaround! I think that should work good enough. People just need to make sure to have that before layer change g-code command.

closed 2020-10-01 by th33xitus