Arksine/moonraker #23: Hostreboot don't work

anjoku, 2020-10-05:
Hello, (please excuse my bad English)

I installed Mainsail on an Orangepi Lite. I used the script from https://github.com/th33xitus/kiauh.
It works so far, only the HOSTREBOOT buttons don't work for me.

The Button HOSTSHUTDOWN work fine.

Where can I set what happens when these buttons are pressed?
Example: HOSTREBOOT should execute "sudo shutdown -h now".

Thank you Andre

[moonraker.log](https://github.com/Arksine/moonraker/files/5327951/moonraker.log)
[klippy.log](https://github.com/Arksine/moonraker/files/5327957/klippy.log)

Arksine, 2020-10-06:
Thanks.  It sounds like your distro does not support the `reboot` command.  I went ahead and pushed an update that executes `sudo shutdown -r now`, which should hopefully work with more distros.

anjoku, 2020-10-10:
Hello, thank you.
I just tried it.

"sudo reboot "

works on the orangePi

Greetings
Andre

KwadFan, 2020-11-19:
Hello @anjoku , 
I discovered this Issue like you.
It seems that the configuration of choosen Modules of MainsailOS and even FluiddPI create this Issue

`export MODULES="base(network, raspicam, klipper_moonraker, canserial, fluidd, mjpgstreamer, password-for-sudo)"`

The Module "password-for-sudo" parses the /etc/sudoers and /etc/sudoers.d to comment out the lines that user pi can "sudo" without any password.
That interferes with the "simple" sudo shutdown -h now or that the API in the back uses.

I am working on a fix for that, till then simply use:
`echo 'pi ALL=(ALL) NOPASSWD: /sbin/shutdown, /sbin/reboot'| sudo EDITOR='tee -a' visudo` 

This will allow pi to use shutdown and reboot command without a need for password. 

Regards
Stephan aka Kwadfan :)

Arksine, 2021-01-03:
Given that this is a disto problem and not a moonraker issue I'm going to close this.  The sudo_fix.sh script is there for CustomPiOS based distros.  Folks using other distros will need to make sure that Moonraker has the proper permissions to execute sudo commands.

closed 2021-01-03 by Arksine