Arksine/moonraker #24: Can't connect with cura's octoprint API connector

Matou25, 2020-10-07:
i want to send file to my printer or print from cura.
Before i used duetRRF to send to DWC2 for clipper.
I changed for moonraker and mainsail.
I open issue in mainsail project, meteyou answer duetrrf can't work with mainsail and we have to use octoprint connector since moonraker emulate octoprint API.

I try to connect with serverIP or serverIP:7125 and API key but impossible to connect.

Do you think it's possible ?

Arksine, 2020-10-07:
It is possible but this is a feature that needs to be implemented in Cura via a plugin.  

Ideally a Cura plugin developer would add native support for Moonraker, but it would be sufficient to use Octoprint's legacy upload API.  Unfortunately Cura uses Octorprint's new upload API, which will not work with Moonraker.

Matou25, 2020-10-08:
thx for your answer

closed 2020-10-08 by Matou25

grigi, 2021-02-27:
> Unfortunately Cura uses Octorprint's new upload API, which will not work with Moonraker.

@Arksine could you please explain why that is? I is just not implemented, or is there some other issue that is getting in the way?

Arksine, 2021-02-27:
Moonraker implements two legacy Octoprint APIs that are necessary to upload files from Slic3r and its derivatives.  These APIs are simple, compatible with Moonraker's upload API, and provide no additional maintenance burden.

Cura's Octoprint plugin requires several APIs go beyond the functionality necessary to upload gcode files (settings, jobs, control, etc).   The plugin queries these APIs and will not work without them.  These APIs are not as simple to implement in Moonraker and they represent a significant maintenance burden should Octoprint change them.

grigi, 2021-02-27:
I am busy with some compatibility code in my fork: https://github.com/Arksine/moonraker/compare/master...grigi:master

As it currently is, uploading of ufp files from Cura is working with the cura-octoprint plugin.

I am just busy finishing up the extra stuff, like the monitoring, and then I'll get to documentation.

I agree that if this significantly changes it may need to be re-done, but at least we can pretend to be a non-upgraded cura instance for a year or so after the breakage.

grigi, 2021-02-28:
I have done a PR here: https://github.com/Arksine/moonraker/pull/102
Still need to do the docs, but the code is ready for review