Arksine/moonraker #25: moonraker crashes at startup

manu7irl, 2020-10-09:
Hi I got elementaryOS running with all the dependencies, on an old atom ITX mainboard.
I think I found a bug or maybe something is missing in my setup... all in all a soon as moonraker service starts it exits, with this error:
```pi@tlm:~$ sudo systemctl status moonraker
[sudo] password for pi:            
● moonraker.service - LSB: moonraker daemon
   Loaded: loaded (/etc/init.d/moonraker; generated)
   Active: active (exited) since Fri 2020-10-09 11:09:57 IDT; 39min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 12966 ExecStop=/etc/init.d/moonraker stop (code=exited, status=0/SUCCESS)
  Process: 12978 ExecStart=/etc/init.d/moonraker start (code=exited, status=0/SUCCESS)

Oct 09 11:09:57 tlm systemd[1]: Starting LSB: moonraker daemon...
Oct 09 11:09:57 tlm moonraker[12978]:  * Starting moonraker moonraker
Oct 09 11:09:57 tlm moonraker[12978]:    ...done.
Oct 09 11:09:57 tlm systemd[1]: Started LSB: moonraker daemon.```

Python version:
```pi@tlm:~$ python --version
Python 2.7.17
pi@tlm:~$ python3 --version
Python 3.6.9
```

Also when I run moonraker manually I got this error:
```pi@tlm:~$ moonraker-env/bin/python3 ~/moonraker/moonraker/moonraker.py 
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 15, in <module>
    import utils
  File "/home/pi/moonraker/moonraker/utils.py", line 11, in <module>
    from queue import SimpleQueue as Queue
ImportError: cannot import name 'SimpleQueue'```

I have no clue what to do from here...
Any help will be appreciated. 
Thanks for your awsome work!
Emanuel.

Arksine, 2020-10-09:
Hi Emmanuel, 

Moonraker requires Python 3.7+.   If you can get that installed on your system it should work.

manu7irl, 2020-10-09:
ok thanks I will try that!

Arksine, 2021-01-03:
Since this discussion seems to have concluded I'm going to close this.

closed 2021-01-03 by Arksine