Arksine/moonraker #28: Service seems active but does nothing.

holmlrs, 2020-10-18:
**Linux distro**
Armbian 20.08.1 Bionic with Linux 5.8.5-sunxi

**Issue**
Everything installs fine. Klipper installs and connects to printer no problem (see klippy.log). However, the moonraker API service does nothing. It installs and starts without problem. But no connection with klipper is made, port 7125 is not listening and the api is not generating a log. 

Any ideas as to how this occurs and how to fix this? 
Thanks in advance!

klippy.log:
[klippy.log](https://github.com/Arksine/moonraker/files/5397306/klippy.log)
printer.cfg (uploaded as txt due to github compatibility):
[printer.txt](https://github.com/Arksine/moonraker/files/5397309/printer.txt)

Arksine, 2020-10-22:
The first thing I would check is the python version.  Moonraker requires Python 3.7.

If your python version is correct, you could try running the application manually to see if an error is sent to stderr:
```
python3 ~/moonraker/moonraker/moonraker.py
````

Makhaira, 2020-11-07:
I may be having a similar issue. Trying to install at the moment and whilst the install script completes successfully I cannot get moonraker to start (no evidence in the klippy.log).

Running the moonraker.py check originally threw up an error with not having tornado so I've added that to python3 and now I can run moonraker.py but nothing happens.

Interestingly in the klippy.log it's reporting my python version as 2.7 even though I have 3.7 installed and have even tried setting a bash alias to redirect python to python3.

Arksine, 2020-11-07:
Klippy requires Python 2, so it will report 2.7 as the verision.  Moonraker requires Python 3.7 +.   If moonraker.py is failing prior to creating /tmp/moonraker.log then it is likely still missing a dependency, running manually should produce an error.  Unfortunately I do not own any devices that run Armbian so I am unable to provide much meaningful help.  If the virtualenv installed correctly, running a slightly different command might be useful.

```
~/moonraker-env/bin/python ~/moonraker/moonraker/moonraker.py
```

Arksine, 2021-01-03:
Since this discussion has concluded I am going to close this.  Given that moonraker now runs as a systemd service it is now possible to possible to use journalctl to view the log as well as any error that may have prevented Moonraker from starting, ie:
```
sudo journalctl -u moonraker.service
```

closed 2021-01-03 by Arksine