Arksine/moonraker #30: Feature Request: Notification Interactions

zervin, 2020-10-21:
Would it be possible to look into bringing some form of notifications to the api. At it's simplest, just a way of getting notifications when a print has completed. In it's most evolved form, notifications at some form of an interval that include a mjpg snapshot if a camera is available. I have several machines happily speeding along, but I have to monitor bookmarked mjpg streams to see when then are finished. One could maybe setup pushover/slack/email/growl..mqtt..whatever makes the most sense with the codebase. Thanks for an amazing contribution!

Arksine, 2020-10-22:
Yes, this is something I can look into and adding a plugin for.  Email notifications would probably be easiest, but I may look into mqtt as well as it seems straight forward to add a "client" and I suspect it would require less user configuration in Moonraker.

zervin, 2020-10-22:
It would end up being on the user side, sorting out what to do with the mqtt messages, but it's a more open standard. I would love to recommend something like pushover/pushbullet/telegram/slack..but I think those tend to add more complexity than warranted. We need you focused on moonraker code, not addressing constant issues from 3rd party api patches!

zellneralex, 2020-11-27:
Here would be a user script that add basic telegram support for moonraker 
https://github.com/Raabi91/moonraker-telegram

zervin, 2020-12-12:
> Here would be a user script that add basic telegram support for moonraker
> https://github.com/Raabi91/moonraker-telegram
I will have to give this a try...

ybjlus, 2021-02-04:
> Yes, this is something I can look into adding a plugin for. Email notifications would probably be easiest, but I may look into mqtt as well as it seems straight forward to add a "client" and I suspect it would require less user configuration in Moonraker.

MQTT would be great! Also for the "power shut down" this way you could use any smart plug or smart device. 

I think if you would broadcast the state of a bunch of variables through MQTT, would make a world of difference. With this you don't have to support different brands/devices/plugins/etc. 
I use home assistant and with this mqtt info, I could make so many things work: notifications (email, phone, brower, tv), auto power off, monitoring and so on. 
Have a look at this: https://github.com/OctoPrint/OctoPrint-MQTT

jrdnlc, 2021-02-21:
Discord notification would probably be easiest to implement

halr9000, 2021-04-10:
Coming from stock marlin and octoprint here, about to jump into klipper land. I am going to miss my telegram and mqtt integrations, so I like the discussion above. I wouldn't use email, personally.

To @zervin point about plugins and clients -- yeah, definitely make it something extensible so others can build on top. I think best bang for the buck would be to do pub/sub to mqtt, and then this would give us home automation capabilities. I even see an mqtt client for telegram, so it might be trivial to do that as well.

Any active work going on in this area?

Arksine, 2021-04-11:
There is a contributor doing some work on a generic notification component.  I believe that initially he plans on supporting configurable HTTP notifications.  When that work is complete I plan to look into adding MQTT support.

eecue, 2021-06-08:
Here's another vote for this! I'm happy to lend a hand.  Where is the notification branch?

pedrolamas, 2021-07-30:
@eecue I believe https://github.com/Arksine/moonraker/pull/184 might be it!

eecue, 2021-07-30:
Awesome! are there other notifications? I just see the MQTT but a generic web hook (or one that's slack flavored) would be awesome. For now I've written this and plan to add web sockets to it: https://github.com/eecue/klipper-slackbot

pedrolamas, 2021-07-30:
That looks quite cool! My personal interest is to use the MQTT status messages to show the printer status on Home Assistant (thus replacing the need to use pulls via REST API as I currently do)

eecue, 2021-07-30:
Yeah I’ll likely do the same with this, but I do want push notifications to slack on print start/stop/errors/etc.

pedrolamas, 2021-07-31:
Actually, that branch was taking a different approach and I believe it is now stale (up to the maintainer if he wishes to refactor it as I do see value in that).

Having said that, I took a stab yesterday in adding print_status changes pushed to MQTT, and it seems to be working ok: https://github.com/Arksine/moonraker/pull/202

jordanruthe, 2021-07-31:
I'm still working on it, but haven't had much time lately. The benefit of a notification module will be granularity in notifications as well as other methods (email, discord, whatever else).

pedrolamas, 2021-07-31:
Thanks @jordanruthe, looking forward for that then! 🙂

soostdijck, 2022-02-20:
I came here looking for notifications (to pushover), is there any news about this? It's been a while since any updates

pataar, 2022-02-23:
I started working on #375. This component utilizes the [apprise](https://github.com/caronc/apprise) library, which includes a ton of notification services.

So eventually you can add this to `moonraker.conf`:
```
[notifier discord]
events: started,completed
url: https://discord.com/api/webhooks/{id}/{secret}}
```

Arksine, 2022-02-23:
#377 has been merged, which should add support for numerous notification backends.  Thanks @pataar!

closed 2022-02-23 by Arksine