Arksine/moonraker #42: missing dependancy

shiftingtech, 2020-11-28:
When I ran ~/moonraker/scripts/install-moonraker.sh -r today on a non-raspbian device (an odroid running ubuntu), the compilation of pillow failed, complaining of "jpeg" not being available.

"The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source."

once I installed  libjpeg-dev, everything installed correctly

I suggest adding libjpeg-dev to the list of packages installed in the apt-get step.

michaelshiel, 2020-12-01:
I had this same issue when installing on armbian running on an Orange PI One. In addition I was also missing zlib1g-dev.

Arksine, 2020-12-01:
I'm open modifying the install script in an attempt to support other distros, at the moment only Raspberry Pi OS based distros are officially supported.   I do not want to install unnecessary dependencies on one platform to support another, so I would need to modify the install script to allow distro specific installations.  As I do not have an Odroid or Orange Pi so I would need someone to test such changes before I merge them.

shiftingtech, 2020-12-01:
at least in the case of zlib1g-dev I don't think you would be installing anything unnecessary: zlib1g-dev is required on all platforms, including Pi OS. It just happens to be in the default image, so it's never made it onto your list.

libjpeg-dev is *slightly* more complicated, since different platforms seem to use different libjpegs.  libjpeg-dev itself seems to be a dummy, that exists only to ensure that the correct actual versions get installed.  So you would *technically* be installing something unnecessary, but it would only be a 60k placeholder that exists exactly for this reason:to abstract the weirdness that is libjpeg...

That all being said, if you did want to make a more selective install script, I would be happy to test.

Arksine, 2021-12-22:
A while back I added zlib1g-dev and libjpeg-dev to the install script, so I am going to close this issue as resolved.

closed 2021-12-22 by Arksine