Arksine/moonraker #44: Panel Babystep bug

robthide37, 2020-12-02:
I had someone else bring this to my attention and i verified on my printer as well. If you baby step with the panel it lifts about 200mm instead of .2 i think their is a missing decimal point or typo somewhere i check line 526 and am trying go over it but im sure you will be able fix way quicker than i ever could lol

Arksine, 2020-12-02:
What version of the PanelDue firmware are you using?

robthide37, 2020-12-02:
1.21.3

robthide37, 2020-12-02:
it worked properly on duet with rrf 2 and 3.01

robthide37, 2020-12-02:
i will try to update panel if you think it helps i just know its a pain to do

Arksine, 2020-12-03:
Moonraker requires v1.24.  Don't update to the new 3.x.x series of PD firmware, as I have not had a chance to look at it however it appears that it may require changes.

Arksine, 2021-01-03:
Since this discussion seems to have concluded I am going to close this.  As mentioned above, the PanelDue plugin requires that firmware version 1.24 to work correctly.

closed 2021-01-03 by Arksine