Arksine/moonraker #46: API: /machine/update/system

KwadFan, 2020-12-05:
After testing the new [update_manager] feature ( Which was for me unclear if it has to set in Moonraker.conf or printer.cfg ),
I discovered that "DEBIAN_FRONTEND=noninteractive" isn't allowed to set.
There for, i will update the sudo_fix.sh, but it has a major issue. It HAVE to be a part of the installation
becaus you have to to update the /etc/environment file, to set DEBIAN_FRONTEND as a fixed System Variable.

Tested on latest FluiddPI Release.
After manual changes it works perfectly.

Who struggels with that Issue can do the following till update:

`echo DEBIAN_FRONTEND=noninteractive >> /etc/environment`

Update your 020-sudo-for-moonraker ( or create file according to your machine ) and append line:

`Defaults!/usr/bin/apt-get env_keep +="DEBIAN_FRONTEND"`

Please make sure you issue an Reboot!

Arksine, 2020-12-05:
Thanks.  I'll clarify the documentation regarding the config.  I'm unsure at the moment as to why `/etc/environment` needs to be written in the install script, however we can discuss it when you submit a PR.  I appreciate your work with this, as I expect in the future most users will prefer to use prebuilt distros rather than vanilla Raspberry Pi OS.

KwadFan, 2020-12-05:
I am working on that too, I am currently fiddling aroun with customPiOS to generate a Distro tailored to Klipper/Moonraker with some comfy Things to choose a Frontend. Indeed this Distro will not need sudo_fix :)
Further more I want completly get rid off "ssh-ing" into the Pi. I only have to understand how to abuse "shell_command.py" and writing a tornado based backend :)
Time will tell...

KwadFan, 2020-12-22:
Fix is merged, so, lets close this :)

closed 2020-12-22 by KwadFan