Arksine/moonraker #51: config_path requirement should not be that it is in home or a subdirectory of home

expeditioneer, 2020-12-19:
Currently the `config_path` must be in **home** or a **subdirectory of home.** This requirement should be relaxed to allow a arbitrary path.

Background: Most Distribution stick to the [FHS](https://refspecs.linuxfoundation.org/FHS_3.0/fhs/index.html). I plan to provide this as a gentoo ebuild and there asubdirectory of **/etc/** should be possible.

jordanruthe, 2020-12-19:
Check out this previous issue: https://github.com/Arksine/moonraker/issues/45

Arksine, 2020-12-20:
Indeed this is similar to #45.  I would be willing to make an exception if this functionality is necessary to package moonraker, however I would require more specificity than any subdirectory in `/etc`. For example, I am willing to allow `/etc/moonraker` as an exception.

expeditioneer, 2020-12-20:
@Arksine feel free to pick a sub directory, - i think `/etc/moonraker` sounds reasonable.

Arksine, 2020-12-20:
Ok, commit 9a309ff allows directories within `/etc/moonraker`.  You can choose `/etc/moonraker` itself if you want, however I would recommend `/etc/moonraker/config` for the configuration files, that way you could specify `/etc/moonraker/gcode_files` for the gcode files if you like.  That is up to you though.  You simply need to make sure moonraker has permisssion to read/write to this directory.

expeditioneer, 2020-12-20:
@Arksine Thank you for your fast implementation

Arksine, 2021-01-03:
Since multiple people have requested a less restrictive directory location I have gone ahead and removed the restriction that folders be located in the HOME directory.  Moonraker will check its permissions on the directory, as long as it has them it will allow it.  The only exception is the system root, which is always prohibited.

closed 2021-01-03 by Arksine