Arksine/moonraker #53: AttributeError: module 'gpiod' has no attribute 'Chip'

mmeulstee, 2020-12-22:
Hello good fellas,

I'm struggling all day to get the power widget shown/activated.
After 3 fresh reinstalls i can't get it to work for some reason.
Maybe there's somebody can watch over this?

[moonraker.log](http://paste.c-net.org/NoddedWhoah)

Arksine, 2020-12-22:
When you reinstalled, did you use the "-r" option?  Or did you simply delete the venv before re-installing?
```
cd ~/moonraker/scripts
./install-moonraker.sh -r
```
If so, it appears that you may have conflicting gpiod package installed on your system.  Please provide the output from the following command:
```
~/moonraker-env/bin/pip list
```
Thanks.

mmeulstee, 2020-12-22:
With reinstall i ment a complete Rpi install
Here's the output of pip list

Package       Version
------------- -------
asn1crypto    0.24.0
configparser  3.5.0b2
cryptography  2.6.1
entrypoints   0.3
enum34        1.1.6
gpiod         1.4.0
ipaddress     1.0.17
keyring       17.1.1
keyrings.alt  3.1.1
pip           20.3.3
pycrypto      2.6.1
PyGObject     3.30.4
pyxdg         0.25
RPi.GPIO      0.7.0
SecretStorage 2.3.1
setuptools    44.1.1
six           1.12.0
virtualenv    15.1.0
wheel         0.36.2

Arksine, 2020-12-22:
Yep, gpiod 1.4.0 is conflicting with the system gpiod package.  FWIW, that python package should not be part of a default RPi OS Lite installation.  I may need to make a change to the install script to prevent this from happening in the future.  In the meantime you can probably remove the package system wide to get going:
```
pip3 uninstall gpiod
```

```
python3 -m

mmeulstee, 2020-12-22:
Thank you very much.

After i did pip3 uninstall gpiod, there still wasnt the power widget.
So thinkin there was still something in the memory, i rebooted the Pi.
And ofcousre there still wasnt a power widget, then i did:
```
cd ~/moonraker/scripts
./install-moonraker.sh -r
```
And its alive', it's alive

Arksine, 2021-01-03:
This issue shouldn't occur with the latest version of moonraker, so I'm going to close this

closed 2021-01-03 by Arksine