Arksine/moonraker #54: Add option to disable loggin to file completely

MagicRB, 2020-12-28:
While logging to stdout is available, moonraker still logs to a file. This is a bit annoying in NixOS as I couldn't find a good place to direct the logfile. When no `-l` is set, moonraker should only log to stdout

Arksine, 2020-12-28:
I won't change the default behavior of `-l` as it would affect all existing installations and the log file is necessary for individuals seeking support.   I'm willing to consider a null value for -l, but I would strongly recommend against packaging moonraker in this manner.  If a user has an issue they will be on their own without a log.

MagicRB, 2020-12-28:
there is no good place to place to really put a log file in NixOS, /var/log is owned by root and I can't really think of any other place

Arksine, 2020-12-29:
Ok.  I'll come up with a way to disable it.

Arksine, 2021-01-01:
This is now possible with the `-n` command line option.

closed 2021-01-03 by Arksine

MagicRB, 2021-01-03:
Thanks! I'll prepare the pull request for the flake and Dockerfile, while I'm here, there is a minor issue with the docker image. I had to bundle klippy+moonraker in one image, as getting UDS working over IP is a major pain. Where should the docker image be? here or in KevinOConnor/klipper?

Arksine, 2021-01-04:
Right, part of the reason for using Unix Socket is to prevent remote clients from connecting directly to Klippy, its for local IPC only.

With regard to the docker image, I think that likely belongs in its own repo.  I see it as something in the category of FluiddPi and MainsailOS.