Arksine/moonraker #56: Allow hardcoding of version at build time

MagicRB, 2020-12-28:
I'd like to be able to specify the version of moonraker via an environment variable or a cmdline option, this would allow me to remove git as a runtime dependency.

Arksine, 2020-12-28:
Moonraker should run correctly without git.  It will report "?" as the current version.  That said, I can look into fetching the version from scripts/version.txt as a fallback option.

Arksine, 2021-12-22:
It is possible to retreive the version from a `.version` file located in the moonraker package.  For example, if you extract the repo to the `moonraker` folder in your home directory, it would need to be located at `/home/pi/moonraker/moonraker/.version`.

closed 2021-12-22 by Arksine