Arksine/moonraker #59: Get rid of the "in home directory" restrictions

MagicRB, 2020-12-31:
For packaging Moonraker + Klippy as a Docker container using Nix it is desirable to not have all the configs and stuff in $HOME. For some directories it's impossible, since the `/share` directory of klippy resides in the Nix store `/nix/store`. Please get rid of this arbitrary restriction or put behind a flag so that I can disable it.

Arksine, 2020-12-31:
This restriction isn't arbitrary, there is a good reason for it.  Moonraker's file manager isn't designed to serve large parts of the file system and allowing configuration in such a manner could lead to unpredictable results. 

That said, this is not the first request of this type I have had.  Previously I made an allowance for "/etc/moonraker", however I am considering other methods to allow configuration outside of the home directory without the potential detrimental affect of infinite recursion when walking the directory.

Arksine, 2021-01-01:
This should be resolved by commit 1814a11.   Moonraker now allows directories outside of the home directory as long as it has the appropriate access permissions.  Only the system root is specifically excluded.

closed 2021-01-03 by Arksine