Arksine/moonraker #61: Feature Request on Power Shutdown

EraserFX, 2021-01-05:
Power on and off via url
I also use the raabi powermanagerscript to control my socket in my Smathome system. This is simply controlled via a url. with a separate on off command. it should be possible to realize that.

Powermanager Script Config:
# printer_off = shell command to switch printer off
printer_off="curl http://IP:PORT/set/loxone.0.15e032e3-02e4-5036-ffffeaed73a38093.active?value=false"
# printer_on = shell command to switch printer on
printer_on="curl http://IP:PORT/set/loxone.0.15e032e3-02e4-5036-ffffeaed73a38093.active?value=true"

alexpthe1, 2021-01-28:
i would apriciate this very much

Arksine, 2021-01-31:
The issue here is that the power plugin must be able to query the device so that it can initialize state.  Simple on/off URLs won't work correctly.

EraserFX, 2021-02-10:
I'm working on a loxone power plugin and close here for now

closed 2021-02-10 by EraserFX