Arksine/moonraker #62: Error "cannot connect printer when add power configuration

oeijoenj, 2021-01-06:
Hi.

I added this script in moonraker.conf :
 [power printer]
 type: gpio
 pin: gpio17
 off_when_shutdown: False
 initial_state: on
and reboot but it show error "No printer connection. Please check moonraker status and / or refresh."

I connect it via SSH and comment those added script, reboot again and it can connect to printer.

Something wrong ?

Arksine, 2021-01-06:
Please attach moonraker.log.   Its possible that you are using a pin that is reserved by another program (sysfs, or possibly Klipper itself).  Moonraker must be able to gain exclusive control over the pin.

oeijoenj, 2021-01-06:
Thanks for your reply.
Please advice, where i can find that file ? Searched all folder but still cannot find it.

Thanks

> Please attach moonraker.log. Its possible that you are using a pin that is reserved by another program (sysfs, or possibly Klipper itself). Moonraker must be able to gain exclusive control over the pin.

Arksine, 2021-01-06:
Thanks.  Unfortunately the log file does not contain your attempt to configure with the [power] item enabled, so I am unable to determine what caused it to fail.  You'll need to reproduce the error, then get the log without rebooting the Pi as a reboot deletes everything in the `/tmp` directory.

Just to verify, you aren't trying to use the Pi's gpio17 in Klipper are you?

oeijoenj, 2021-01-06:
I dont know whether I did reproduce the correctly but this the log I got before I reboot it.
Regarding to GPIO 17, yes, I intend to use that pin, the same pin I use when I use octoprint before moved to fluidd.
Any concern? I dont mind to change to other pin if needed.

thanks so much

[moonraker.log](https://github.com/Arksine/moonraker/files/5775653/moonraker.log)
 

> Thanks. Unfortunately the log file does not contain your attempt to configure with the [power] item enabled, so I am unable to determine what caused it to fail. You'll need to reproduce the error, [...]

Arksine, 2021-01-06:
Unfortunately the [power] plugin is not configured in that log either.  You will need to add the [power printer] section to mooraker.conf, restart moonraker, then grab the log before you reboot the Pi.  If moonraker is unable to start you may have to use sftp to do this.

If any other application, such as Octoprint or Klipper, currently has access to gpio17 then Moonraker will be unable to acquire it, this will result in an error.   You need to make sure that Klipper isn't using the pin in printer.cfg.  If you would like you can also upload klippy.log as well so I can see what your klippy configuration looks like.

oeijoenj, 2021-01-06:
Yes, moonraker is unable to start then i cannot access printerIPaddress/server/file/moonraker.log.
Did SFTP it but it cannot access that folder. Would you assist me where I can find that log using SFTP?

btw, regarding to GPIO17. Yes, I was using it in Octopi but it use another micro SD so now only has fluidd in raspberry.

Arksine, 2021-01-06:
Ok, you need to reinstall moonraker.  This must be an older installation that never installed the gpiod module dependency.

SSH into the Pi and enter the following commands:
```
cd ~/moonraker
./scripts/install-moonraker.sh -r -f -c /path/to/moonraker.conf
```
Replace `/path/to/moonraker.conf` with the actual path of where the file is located.  If moonraker.conf is just in the home directory you can leave it out, ie:
```
./scripts/install-moonraker.sh -r -f
```

oeijoenj, 2021-01-06:
Hi. Thanks for your response.

Just did script ./scripts/install-moonraker.sh -r -f -c /home/pi/klipper_config/moonraker.conf and installed successfully but I found that there is newer version of moonraker available found in UI setting, screenshot attached.
Everything looks good and I uncomment the power section and it freeze again.

Before I send this message, I did update moonraker and it cannot update. Only box with Repo moonraker: Updating Repo... and nothing happen.

Thanks

[moonraker.log](https://github.com/Arksine/moonraker/files/5778737/moonraker.log)

[klippy.log](https://github.com/Arksine/moonraker/files/5778736/klippy.log)

[image]

Arksine, 2021-01-07:
The log is showing the power plugin has successfully loaded.   With regard to the update, the log doesn't seem to include that request from Fluidd.  Sometimes the "git" lock file doesn't get removed properly, however this should result in an error.

SSH into the pi and try to do a manual git pull:
```
cd ~/moonraker
sudo service moonraker stop
git pull
sudo service moonraker start
```

If you get an error saying that `index.lock` exists, you may need to manually remove it:
```
rm -f ~/moonraker/.git/index.lock
```
Then you should be able to pull and restart the service.

oeijoenj, 2021-01-07:
Hi. Thanks so much.
Work very well now and even better you add tab in dashboard to toggle on and off.
Very great work. Appreciate it

[image]

closed 2021-01-07 by oeijoenj

dushyantahuja, 2021-01-07:
was setting up my printer today with a new version of fluiddpi - and looks like the version of moonraker bundled with fluiddpi doesn't have the power features, not sure why...

SmoothieAq, 2021-01-10:
I have just started with fluiddpi. I had to update Moonraker, but after that Moonraker starts fine with the [power printer] section. However, nothing appears in Fluidd. (sorry, perhaps I ought to report this in the Fluidd rep, but it seems same topics as this issue).

SmoothieAq, 2021-01-10:
> ```
> cd ~/moonraker
> ./scripts/install-moonraker.sh -r -f -c /path/to/moonraker.conf
> ```
ok, the pull was up to date, but the install script fixed it. Sorry I didn't read before writing...
there seems to be some problem with the current fluiddpi and the update process, that causes the manual install to be needed.

Arksine, 2021-01-10:
Thanks @SmoothieAq  for the update.

@dushyantahuja It sounds like fluiddpi was built against an older version of moonraker.  Its possible that it didn't have the necessary dependencies required to load the power module (specifically it needs gpiod).  I would recommend trying the re-install as mentioned above.  Remember to put the actual path to moonraker.conf for the -c option.  I'm guessing it is at `/home/pi/klipper_config/moonraker.conf`, but you will want to check your file system to be sure.

SmoothieAq, 2021-01-10:
Yes it is `/home/pi/klipper_config/moonraker.conf`
And you can not use a relative path ;-)

dushyantahuja, 2021-01-11:
Thanks Eric, I was able to re-install Moonraker manually, however wanted to
report this issue for fluiddpi.

Regards