Arksine/moonraker #64: FR: Have sent gcode commands saved in console history

cadriel, 2021-01-11:
Currently, when you refresh a client - the sent command history is lost. This can be a bit confusing for some users as the console history looks a bit different after refresh.

It'd be be nice if we had an option to retain this.

cadriel, 2021-01-11:
Secondarily to this - if we could have a clear console endpoint too, that would be a nice to have.

Arksine, 2021-01-12:
Ok, I'm  working on adding this, however I think I would like  some clarification on the clear console endpoint.  Right now we have the gcode store endpoint and it returns something like this:
```python
{'gcode_store': [{msg1, time1}, {msg2, time2}, ...]}
```
Ideally I would simply add another field to that endpoint so it looks like this:
```python
{
  'gcode_store': [{msg1, time1}, {msg2, time2}, ...],
  'command_store':[{cmd1,time1}, {cmd2, time2}]
}
```
I could add another endpoint, but I'm unsure if its really necessary.

Arksine, 2021-01-13:
Closing this as support was added in d70bd8d

closed 2021-01-13 by Arksine