Arksine/moonraker #66: FR: Add Klipper's PWM value to /server/temperature_store

CapnBry, 2021-01-11:
When I warm up my printer, I like to wait long enough for the heated bed to soak up to temperature. It takes a while for a thick MIC6 plate to actually stabilize at print temperature, long after the thermistor reports it first gets there. To do this I usually tail the klippy.log with a python regex and watch the "pwm=" in the log and wait for that to stabilize, meaning the bed has reached equilibrium. To make this more sophisticated, I'd like to get this into some webui somewhere. Klipper already exports this information through its API, so moonraker is the next piece of software that needs to expose it.

Klipper exports it as "power" in its API, so it was trivial to add to your excellent framework as "power" to the endpoint `/server/temperature_store`.
```
diff --git a/moonraker/plugins/data_store.py b/moonraker/plugins/data_store.py
index 77760a7..4a65a0a 100644
--- a/moonraker/plugins/data_store.py
+++ b/moonraker/plugins/data_store.py
@@ -66,9 +66,10 @@ class DataStore:
                 else:
                     new_store[sensor] = {
                         'temperatures': deque(maxlen=TEMPERATURE_STORE_SIZE),
-                        'targets': deque(maxlen=TEMPERATURE_STORE_SIZE)}
+                        'targets': deque(maxlen=TEMPERATURE_STORE_SIZE),
+                        'powers': deque(maxlen=TEMPERATURE_STORE_SIZE)}
                 if sensor not in self.last_temps:
-                    self.last_temps[sensor] = (0., 0.)
+                    self.last_temps[sensor] = (0., 0., 0.)
             self.temperature_store = new_store
             # Prune unconfigured sensors in self.last_temps
             for sensor in list(self.last_temps.keys()):
@@ -86,17 +87,19 @@ class DataStore:
     def _set_current_temps(self, data):
         for sensor in self.temperature_store:
             if sensor in data:
-                last_temp, last_target = self.last_temps[sensor]
+                last_temp, last_target, last_power  = self.last_temps[sensor]
                 self.last_temps[sensor] = (
                     round(data[sensor].get('temperature', last_temp), 2),
-                    data[sensor].get('target', last_target))
+                    data[sensor].get('target', last_target),
+                    data[sensor].get('power', last_power))

     def _update_temperature_store(self):
         # XXX - If klippy is not connected, set values to zero
         # as they are unknown?
-        for sensor, (temp, target) in self.last_temps.items():
+        for sensor, (temp, target, power) in self.last_temps.items():
             self.temperature_store[sensor]['temperatures'].append(temp)
             self.temperature_store[sensor]['targets'].append(target)
+            self.temperature_store[sensor]['powers'].append(power)

     async def _handle_temp_store_request(self, web_request):
         store = {}
```

I can make a pull request for this change if you'd like but I wanted to make sure you would be willing to accept it before I went through the effort of forking the repo and all that. It only changes the one file too, so I wouldn't mind if you just applied the patch and committed it yourself either. Just let me know if this sounds like something we can add.

Arksine, 2021-01-11:
I'm ok with this addition, feel free to make a pull request.  I think keeping the PWM values as floats makes sense as it will match what clients receive when they are subscribed to a particular sensor.

CapnBry, 2021-01-11:
Sounds great, thanks! I'll make a PR for it later tonight or in the morning.

Arksine, 2021-01-13:
I'm going to go ahead and close this as it was resolved by #68

closed 2021-01-13 by Arksine

dewi-ny-je, 2021-01-20:
@CapnBry how do you plot that value now that the pull request has been merged? I guess you use Mainsail or Fluidd.
Thanks

CapnBry, 2021-01-20:
I just made a little local webpage to test it and display the value. Will work with the frontend developers individually to figure out how they'd like to present the information. Still experimenting with that myself though currently using KlipperScreen.