Arksine/moonraker #69: Feature: Tornado WebSocket as Unix Domain Socket

expeditioneer, 2021-01-14:
Tornado has the capability to 'export' the functionality also as Unix Domain Socket.

Normally moonraker is run behind a web server like NGINX or Apache HTTP. 

This has serveral benifits, e.g:
 - increased security
 - get rid of unnecessary tcp frame

Arksine, 2021-01-24:
I think that if a client wishes to connect via Unix Domain Socket it would be better off connecting straight to Klipper. Moonraker is generally intended to run as a web server, so hosting a Unix Domain Socket is a bit out of its scope.  If a local application wants access it can still connect to the Websocket.

Arksine, 2022-11-19:
Something similar is now available.  Moonraker now serves the JSON-RPC API over a Unix Socket, however it doesn't "export" the websocket protocol.  It uses a protocol similar to that of Klipper's unix socket, where the encoded json is terminated with an ETX character.

closed 2022-11-19 by Arksine