Arksine/moonraker #70: Feature: Add support esphome api for power plug

tayanov, 2021-01-21:
Please add support esphome api for power plug.
If you show home assistant config and use tasmota power, may be you can add support [power printer] and [power device_name] with esphome restapi ?
[https://esphome.io/web-api/index.html](url)
I use sonoff pow2 on esphome. i see powermeter and can turnon off printer from home assistant, but not from Fluiid or other klippers web gui.

tayanov, 2021-01-27:
or may be add simple pull http url e.g. wget  as [power printer]

alexpthe1, 2021-01-28:
or just the possibility to run a shell script

Arksine, 2021-01-31:
As mentioned in issue 61, simple url's won't work with the power plugin, as it must be able to query the state of the device.   The tasmota device was contributed by a user.   Someone who owns one of "esphome" devices is free to contribute an implementation.  

There are major security implications with running configurable scripts, so it is unlikely that I add that ability.

tayanov, 2021-02-03:
> As mentioned in issue 61, simple url's won't work with the power plugin, as it must be able to query the state of the device. The tasmota device was contributed by a user. Someone who owns a "esphome" device is free to contribute an implementation.
> 
> There are major security implications with running configurable scripts, so it is unlikely that I add that ability.

ok, im migrate to tasmota, but its not work, with any config i see that

Fluidd warnings found.
Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
power
Moonraker plugin configuration can be found here.

its my config
[power printer]
type: tasmota
address: 10.0.0.100
output_id: 1

i try anything config but hope is die)

Arksine, 2021-02-03:
Moonraker.log should tell you what the error is.  I suspect it is because device query failed, but I won't be able to tell without the log file.  If that is the case, you need to make sure that `SetOption26` is set in Tasmota.

tayanov, 2021-02-04:
[moonraker (3).log](https://github.com/Arksine/moonraker/files/5924056/moonraker.3.log)
i cant see any error at log

tayanov, 2021-02-04:
i comment 13 line at power.py #import gpiod
and all works. its trouble as #75

why venv not install gpiod module?

Arksine, 2021-02-04:
You can't power on with a GCode Macro as Klipper can only execute RESTART and FIRMWARE_RESTART when it is in the shutdown state.  You must use the API to power on.

iJebus, 2023-07-05:
Not to go resurrecting a dead issue, but on the basis of the title, I think this can be closed. I'm controlling an ESPHome power plug using the generic HTTP config, e.g.

```
[power SV06]
locked_while_printing: True
restart_klipper_when_powered: True
type: http
on_url: http://x.x.x.x/switch/athom_smart_plug_v2/turn_on
off_url: http://x.x.x.x/switch/athom_smart_plug_v2/turn_off
status_url: http://x.x.x.x/switch/athom_smart_plug_v2
request_template:
  {% if command in ["on", "off"] %}
    {% do http_request.set_method("POST") %}
    {% do http_request.set_body({}) %}
  {% endif %}
  {% do http_request.send() %}
response_template:
  {% if command in ["on", "off"] %}
    {% do async_sleep(1.0) %}
    {% do http_request.set_method("GET") %}
    {% do http_request.set_body(None) %}
    {% do http_request.set_url(urls.status) %}
    {% set response = http_request.send() %}
    {% do response.raise_for_status() %}
  {% endif %}
  {% set resp = http_request.last_response().json() %}
  {resp["state"]}
```