Arksine/moonraker #71: Feature request: Power on gcode

wormiedk, 2021-01-21:
My printer uses UART communication with the drivers which means that klipper has to be restarted after a power cycle. Further the printer is not powered from USB so the port dissapears when the printer is off.

It would be great if this could be included in the startup code from my slicer, e.g. by using a macro. I tried the following macro:

[gcode_macro POWER_ON_PRINTER]
gcode:
  {action_call_remote_method("set_device_power",
                             device="printer",
                             state="on")}
  RESTART

However when I try issuing the command I get a response that moonraker cannot connect to klippy and therefrore neither fluid or moonsail allows me to issue the command.

Arksine, 2021-01-24:
This isn't possible.  Klipper cannot execute gcode macros when it is in the shutdown state, this is part of the reason why Moonraker has a "power" plugin.  You will have to toggle power from Fluidd.

wormiedk, 2021-01-24:
Would it be possible for the power plug-in to issue a restart command after a short delay when the printer is powered on?

Arksine, 2021-01-24:
It could perhaps be made an option.  I'll look into it.

wormiedk, 2021-01-24:
That would be great. Maybe also a shutdown command :)

Arksine, 2021-01-24:
Klipper will automatically enter shutdown when the mcu is disconnected

dushyantahuja, 2021-01-25:
This would be really helpful - firmware_restart is required after switching
on the printer to ensure the TMC drivers are initialised properly.

I believe klipperscreen does something similar.

Regards

Arksine, 2021-01-27:
It is now possible to configure a `[power]` device to perform a FIRMWARE_RESTART after power on.  See the `restart_klipper_when_powered` and `restart_delay` options in [configuration.md](https://github.com/Arksine/moonraker/blob/master/docs/configuration.md#power).

closed 2021-01-27 by Arksine

dushyantahuja, 2021-01-27:
That's great. Thanks a lot.