Arksine/moonraker #73: Moonraker Failed Plugins: [power] after latest update

pheonix3330, 2021-01-22:
moonraker log attached. 

Looks like the issue is here:

2021-01-22 22:02:40,672 [power.py:__init__()] - Power plugin loading devices: ['power tasmota_plug']
2021-01-22 22:02:40,672 [moonraker.py:load_plugin()] - Unable to load plugin (power)
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 148, in load_plugin
    plugin = load_func(config)
  File "/home/pi/moonraker/moonraker/plugins/power.py", line 439, in load_plugin_multi
    return PrinterPower(config)
  File "/home/pi/moonraker/moonraker/plugins/power.py", line 39, in __init__
    self.devices[dev.get_name()] = dev
  File "/home/pi/moonraker/moonraker/plugins/power.py", line 174, in get_name
    return self.name
AttributeError: 'Tasmota' object has no attribute 'name'
[moonraker (1).log](https://github.com/Arksine/moonraker/files/5858919/moonraker.1.log)

My tasmota plug is configured as follows:

[power tasmota_plug]
type = tasmota
address = 192.168.86.20
off_when_shutdown = True

Any idea what the issue could be? Was working before the latest moonraker update.

Arksine, 2021-01-22:
Thanks.  I just pushed a fix, it should work now.

pheonix3330, 2021-01-22:
Thanks, working now!

closed 2021-01-22 by pheonix3330