Arksine/moonraker #74: Feature request: Power off certain minutes after print finished

oeijoenj, 2021-01-24:
Hi,

Is it possible to power off printer certain minutes after print finished ?
thanks

jonathan

Arksine, 2021-01-24:
You could set up a [delayed_gcode] that powers off your printer, to be called in your print's end gcode.  Assuming the gpio is named "printer", you could add the following to printer.cfg:
```
[gcode_macro POWER_OFF_PRINTER]
gcode:
  {action_call_remote_method("set_device_power",
                             device="printer",
                             state="off")}

[delayed_gcode delayed_printer_off]
initial_duration: 0.
gcode:
    POWER_OFF_PRINTER
```

Then in your  end gcode add something like this:
```
UPDATE_DELAYED_GCODE ID=delayed_printer_off DURATION=120
```
That would power down your printer 2 minutes after the print completes.

oeijoenj, 2021-01-24:
Hi Arksine,

Thanks for your response.
Actually i use this script in moonraker.conf 

`[power printer]
type: gpio
pin: !gpio22
off_when_shutdown: False
initial_state: on`

Any idea how to implement delayed gcode ? I dont know much how to do scripting.
Thanks so much.

> You could set up a [delayed_gcode] that powers off your printer, to be called in your print's end gcode. Assuming the gpio is named "printer", you could add the following to printer.cfg:
>
> ``` [...]

Arksine, 2021-01-24:
My previous comment outlined how to implement the delayed gcode in printer.cfg.

oeijoenj, 2021-01-24:
Got it.
Just did the change and works as expected.
Thanks so much.

> My previous comment outlined how to implement the delayed gcode in printer.cfg.

closed 2021-01-24 by oeijoenj