Arksine/moonraker #75: GPIO Power not working

leadgtr7, 2021-01-24:
I suspect this has to do with #73 but i don't know

my code is copied pasted from the example on how to set this up but still doesn't work. My version is v0.2.8-37 

Here is the code

[power printer]
type: gpio
pin: gpio26
off_when_shutdown: True
initial_state: off

attached is my log file
[moonraker.log](https://github.com/Arksine/moonraker/files/5861601/moonraker.log)

jordanruthe, 2021-01-24:
According to your log file, something is wrong with gpiod on your installation. I'd recommend re-running the moonraker installation script and it should fix any issues by: `./install-moonraker.sh -r` from the moonraker/scripts directory

Arksine, 2021-01-24:
MainsailOS (and likely its deriviatives) currently doesn't symlink the `gpiod` dependency.  I believe that this will be fixed in the next release.

As Jordan mentioned you can fix this by running the install script.

leadgtr7, 2021-01-24:
Thanks for the tip on that! it works!

Just wanted to add for anyone who might look at this in the future to make sure to reboot as it didn't work before that

closed 2021-01-24 by leadgtr7