Arksine/moonraker #76: moonraker pointing to python 3.7 deps instead of 3.8

turtiustrek, 2021-01-24:
Hello, i have python 3.8.3 installed and when using update_manager, it fails.
```
2021-01-25 02:23:30,369 [moonraker.py:load_plugin()] - Unable to load plugin (update_manager)
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 148, in load_plugin
    plugin = load_func(config)
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 942, in load_plugin
    return UpdateManager(config)
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 51, in __init__
    "moonraker": GitUpdater(self, mooncfg, MOONRAKER_PATH, env)
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 453, in __init__
    raise config.error("Invalid path for option '%s': %s"
confighelper.ConfigError: Invalid path for option '/home/pi/moonraker-env/lib/python3.7/site-packages': env_package_path
```

```
pi@orangepioneplus:~$ ls  /home/pi/moonraker-env/lib/
python3.8
```
Thanks,
turtius

gdachs, 2021-01-24:
I have the same issue. The missing package is installed but in /home/gdachs/moonraker-env/lib/python3.8/site-packages.

turtiustrek, 2021-01-25:
> Okay, I fixed it for me by adding a symlink instead.

symlink to what exactly? The directory to the python3.8 folder or  `update_manager.conf`.

gdachs, 2021-01-25:
`gdachs@x5sapro:~/moonraker-env/lib$ ls  -l
insgesamt 4
lrwxrwxrwx 1 gdachs gdachs    9 Jan 25 00:07 python3.7 -> python3.8
drwxrwxr-x 3 gdachs gdachs 4096 Jan 25 00:07 python3.8`

menelike, 2021-02-03:
As a workaround, you can align the path in https://github.com/Arksine/moonraker/blob/943b78195bc44b67206246f2e5b6af7bba5f407d/scripts/update_manager.conf#L10 to `python3.8` (or `python3.9` in my case). I think `env_package_path` should ideally be generated dynamically.

and you gotta deal with the nasty message in `fluidd` if you're a perfectionist:

[image]

gdachs, 2021-02-03:
> As a workaround, you can align the path in
>
> https://github.com/Arksine/moonraker/blob/943b78195bc44b67206246f2e5b6af7bba5f407d/scripts/update_manager.conf#L10
>
> to `python3.8` (or `python3.9` in my [...]

This is exactly what I described in my first post.

Arksine, 2021-02-03:
FWIW, I should have a resolution for this issue pushed soon.  It is part of a series of patches I plan to push, one of which clients need to adapt before I do so.  When they are ready this should work without a symlink.

dimalo, 2021-02-03:
For the perfectionists:
I've just [updated my docker image](https://github.com/dimalo/klipper-web-control-docker/blob/main/klipper/Dockerfile#L81) of klipper/moonraker to get rid of the message:

Assuming you have python 3.9 (moonraker venv as well) and you don't have it installed under /usr/lib/python3, you have to issue two commands to symlink an existing python3.9 in /usr/local/lib:
'''bash
mkdir -p /usr/lib/python3 && ln -s /usr/local/lib/python3.9/site-packages /usr/lib/python3/dist-packages
mkdir -p ${HOME}/moonraker-env/lib/ && ln -s ${HOME}/moonraker-env/lib/python3.9 ${HOME}/moonraker-env/lib/python3.7
'''

now moonraker doesn't feel dirty anymore...

closed 2021-02-06 by Arksine

Arksine, 2021-02-06:
This issue has been fixed, however you all will need to remove your symlinks for it to work correctly.  It uses Python's "glob" package to determine the correct directory for the `site-packages`, however if glob returns more than one directory moonraker won't know which one to use.  It sees the symlink as an additional directory, so that will confuse it.

knoopx, 2021-07-02:
related to this is `python_dist_path` which points to `/usr/lib/python3/dist-packages`. I'm not using debian and `dist-packages` is a debian-specific convention. There's currently now way of disabling it other than modifing `update_manager.conf` and getting dirty status.

Arksine, 2021-07-02:
I'm currently working on significant changes to the update manager and part of it actually removes the `dist_path`.  The bad news is that instead of symlinking the gpiod module in the virtualenv I instead append the dist path to sys.path() before importing it in the power module.  If you don't use Moonraker to toggle gpios then this won't be an issue for you, otherwise we may need to hardcode different locations to test for the gpiod module.  Sadly we cant use `site.getsitepackages()` due to a limitation set by the virutalenv.

That said, officially the update manager only supports debian and its derivatives as this is what I dev on and test against.  I'm not opposed to supporting other distros in the future and would be happy to accept PRs toward this goal.  This is something that should definitely wait until my currrent changes are merged as the refactor the the update manager is fairly significant (it adds a beta channel that deploys Moonraker and Klipper in a zipped file).

commit fe2cec10 (2021-02-06): update_manager: use glob pattern matching to find site package folder

This allows for python version flexibility.  Resolves #76.