Arksine/moonraker #77: Error trying restart klipper or moonraker with api

Deymos42, 2021-01-28:
Hi,
when I run POST / machine / services / restart? service = <klipper> I get the following error: 

Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/home/pi/moonraker/moonraker/app.py", line 304, in post
    await self._process_http_request('POST')
  File "/home/pi/moonraker/moonraker/app.py", line 319, in _process_http_request
    WebRequest(self.request.path, args, method, conn=conn))
  File "/home/pi/moonraker/moonraker/plugins/machine.py", line 52, in _handle_service_restart
    raise self.sever.error(
AttributeError: 'Machine' object has no attribute 'sever'

Any ideas? thanks

Arksine, 2021-01-28:
Two things:
1) You did indeed find a typo that I need to correct.
2) The typo is in an error handler.  You aren't sending a valid service name in the query string.  For example to restart Klipper, you need to send the following command:\
`POST /machine/services/restart?service=klipper`

Deymos42, 2021-01-28:
Ou shit, I didn't try to send the command without the <>. My bad. Thanks, works perfecly!

closed 2021-01-28 by Deymos42