Arksine/moonraker #78: Error restarting klipper

Deymos42, 2021-01-28:
Hi, I accidentally closed the last issue sry, the command: / machine / services / restart? Service = klipper, returns ok but klipper does not restart. Could it be because of the typographical error of "sever"?

Arksine, 2021-01-28:
The typo has been fixed and it would not prevent the service from restarting if the endpoint was provided with valid parameters.  I would need the log to help troubleshoot further.  

Keep in mind that the URL is case sensitive.

Deymos42, 2021-01-29:
Hi, the last 2 lines of .log show that the command was executed correctly:

2021-01-29 11:50:48,315 [shell_command.py:run()] - Command (sudo systemctl restart klipper) exited with return code 1
2021-01-29 11:50:48,318 [web.py:log_request()] - 200 POST /machine/services/restart?service=klipper (192.168.0.119) 129.23ms

If I manually run sudo systemctl restart klipper it works fine. But with moonraker klipper does not restart. I have given all the permissions to the moonraker scripts folder (chmod 777) and nothing.

I attach the complete log. Any ideas?

[moonraker.log](https://github.com/Arksine/moonraker/files/5893680/moonraker.log)

Arksine, 2021-01-29:
It is a sudo permission issue.   If you are not running MainsailOS, FluiddPi or Raspberry Pi OS  you will need to determine what permissions are necessary on your distro to execute systemctl from Moonraker.

Deymos42, 2021-01-29:
I'm running moonraker on Raspberry Pi OS, it seems that running the script `sudo_fix.sh` inside `/moonraker /scripts` solves it.

Arksine, 2021-01-29:
Interesting.  Stock RPi OS shouldn't need the script, but I'm glad it resolved your issue.

Deymos42, 2021-01-29:
Well, it's not Stock RPi OS, it's an octoprint image. But use this distribution: Raspbian GNU / Linux 10 (buster). thanks for the replies. 
Regards

closed 2021-01-29 by Deymos42