Arksine/moonraker #81: Feature Request: UFP parsing/unzipping from filesystem

Rejdukien, 2021-01-31:
Currently UFP files are being parsed only on file upload. However, sometimes it might be more practical to just use good old samba network shares to copy files to the raspberry - such as currently curas octoprint upload plugin using a different api version than implemented in moonraker (or atleast that's the current situation, if I understood correctly).

Therefore I'd like to propose running any detected .ufp file in the gcodes folder through the same method as used on file upload - Unzipping the UFP, extracting thumbs/gcode from it and discarding the ufp.

https://github.com/Arksine/moonraker/blob/943b78195bc44b67206246f2e5b6af7bba5f407d/moonraker/plugins/file_manager.py#L421

I've had semi-success by just naively putting something like the following in the get_file_list method, but that's a rather dirty solution and just issuing "server.files.get_directory" via websocket won't actually call that method.

```python
if root == 'gcodes' and ext == '.ufp':
    fullpath = os.path.join(dir_path, name)
    file = open(fullpath, mode='rb')
    self._unzip_ufp({'full_path':fullpath, 'filename':name, 'body':file.read()})

DruckiMcDruck, 2021-01-31:
I just ran into this exact issue. I'm using Cura 4.8.0 and Moonraker + Mainsail. As the Cura Octoprint Plugin doesn't work with Moonraker, i setup a samba netshare to save files directly onto the pi. Sadly moonraker doesn't process these files at all, leaving me without preview.

Arksine, 2021-01-31:
Generally speaking it is intended for users to manage files via the interface.  While its possible to monitor folders, there are some caveats and it can cause real problems depending on how the system is configured.   In the future I may look into an event driven solution where the file manager is notified by the kernel, but that is low priority so I have no timetable on it.  I am however open unzipping a UFP when a directory is refreshed in the UI, similar to the OPs example.

In an ideal world someone will add a Cura plugin with support for Moonraker.  This could be done via direct API support or by using the limited Octoprint APIs Moonraker supports for file uploads.  I don't use Cura nor do I have time to maintain such a plugin so it would need to be user contributed.

commit 96486303 (2021-02-06): file_manager: process ufp files found when refreshing a directory

This allows users to manaully place ufp files in the gcode directory.  Resolves #81.
closed 2021-02-06 by Arksine