Arksine/moonraker #82: DEBIAN10 - Fluidd warnings on update manager

kintarojs, 2021-02-04:
I created two instances of klipper + moonraker using Kiauh script : 
https://github.com/th33xitus/KIAUH
In Fluidd webclient, both instances are connected to my two printers but raise a warning : 

> Fluidd warnings found.
> Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
> update_manager
> 

Plus, when I tried to heat my extruder, it seems that one 'printer session' entered in a loop getting temperatures alternatively from both printers, making strange charts.

here my two moonraker.cfg
`[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: True
config_path: /home/jsh/klipper_config
klippy_uds_address: /tmp/klippy_uds

[authorization]
enabled: True
api_key_file: ~/.moonraker_api_key
trusted_clients:
    127.0.0.1
    192.168.0.0/16
    192.168.1.0/24
    ::1/128
    FE80::/10
cors_domains:
    http://*.local
    http://my.mainsail.xyz
    https://my.mainsail.xyz
    http://app.fluidd.xyz
    https://app.fluidd.xyz
    http://192.168.1.48
    http://192.168.1.48:*

[update_manager]

[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

#[update_manager client KlipperScreen]
#type: git_repo
#path: /home//home/jsh/KlipperScreen
#origin: https://github.com/jordanruthe/KlipperScreen.git
#env: /home//home/jsh/.KlipperScreen-env/bin/python
#requirements: scripts/KlipperScreen-requirements.txt
#install_script: scripts/KlipperScreen-install.sh
`

Following one is complete, I removed all update_manager in order to try to remove the warning fro fluidd. With no success...
`[server]
host: 0.0.0.0
port: 7126
enable_debug_logging: True
config_path: /home/jsh/klipper-2_config
klippy_uds_address: /tmp/klippy-2_uds

[authorization]
enabled: True
api_key_file: ~/.moonraker_api_key
trusted_clients:
    127.0.0.1
    192.168.0.0/16
    192.168.1.0/24
    ::1/128
    FE80::/10
cors_domains:
    http://*.local
    http://my.mainsail.xyz
    https://my.mainsail.xyz
    http://app.fluidd.xyz
    https://app.fluidd.xyz
    http://192.168.1.48
    http://192.168.1.48:*

[update_manager]
`

Arksine, 2021-02-05:
While multi-seat installations are possible they aren't officially supported, so I won't be much help for any issues you may face there.  I couldn't begin to explain why one session is receiving temperature data from another, however it sounds like its either a client issue or system configuration issue.  Since KIAUH supports installing multiseat the Klipper discord might be a good place to ask questions as there may be others running it.

With regard to the warning, the specific error that caused the update manager to fail should be available in moonraker.log.   With that I could help you troubleshoot further.

cadriel, 2021-02-20:
There was a bug in Fluidd that was fixed with 1.7 - prior to that, version 1.6 had an issue with cross contamination of data if you added more than one instance to the host.

There's also scenario's where you can configure things in an odd way causing the same issue.

@kintarojs Is this still an issue?

Arksine, 2021-04-21:
Since there is no activity on this issue I am going to close it.  As mentioned above, I tend to discourage multi-seat installations as they can result in a variety of potential issues.

closed 2021-04-21 by Arksine