Arksine/moonraker #83: Power on a device with restart_klipper_when_powered: True fails sometimes

zellneralex, 2021-02-05:
Hi, first of all I am not sure if I should call it Bug or Feature Request.

HW Setup:  PI GPIO Relay to interrupt 24V AC in. 5V for the PI is always on mcu is not powered by 5V USB 
SW Setup:  `[power psu]
type: gpio
pin: !gpio14
initial_state: on
off_when_shutdown: True
locked_while_printing: True
restart_klipper_when_powered: True
restart_delay: 10
`
I see in rare cases 1 out 10-15 that the intended FW-Restart did not get klipper in a healthy state again.

So I try to find the root cause, here the findings:
Switching off the 24V Power some  time get the klipper service in the state "active (exited)" and in this cases a FW-Restart will not work to get klipper up and running. I did not find a root case why that happens sometimes. You can clearly different the situations according the error messages shown in mainsail. Normal failure like "can not communicate to mcu" vs  "moonraker can not connect error because of missing klipper uds"

I am aware that switching off 24V is a hard event for klipper so I do not want blame klipper for that. My impression was that the moonraker power parameter  restart_klipper_when_powered was added exactly for that situation (restart klipper after switching 24V on again). A manual Restart of the klipper Service (either per ssh or using mainsail) always fixed the situation

Possible Solution: Either add  a klipper restart service event before before doing the FW-Restart or replace the FW-Restart with that. If you tell me one of these solution is viable I can try to integrate it in my power.py (python is not my world) and test it out in my setup
Workaround: I resolved the situation by adding a udev rule to restart klipper whenever a SKR1.4 USB device is detected but that will not work at boards where you can not split 5V onboard from 5V USB (like the SKR mini E3)

Arksine, 2021-02-06:
I would prefer not to add a service restart to the power up sequence as there are timing issues at play. I could perhaps let one choose between "firmware_restart" and restarting the service, but I'm unsure I could execute both reliably.  Also, like your udev rule, its a hack or workaround and doesn't address the underlying issue.

I think it might be worthwhile to determine what is killing the Klipper service when you power off your device.   Most likely the kernel is sending it a signal, perhaps its SIGTERM, SIGINT, or SIGHUP.  Can or should that be handled in Klipper?

zellneralex, 2021-02-06:
Honestly it is hard to reproduce I tried it once 50 times in a row and never get the klipper service down. Then my idle timeout triggered the 24PS down in the night and at the next morning the service was down

zellneralex, 2021-02-06:
I am willing to help debugging it but would need advice how to. My main coding knowledge is chip design so more Verilog or SystemVerilog and not that much Python.

Arksine, 2021-02-06:
The next time the service dies enter the following command before restarting it:
```
sudo systemctl status klipper
```
Screencap the result and upload to this thread.   Hopefully it will contain the signal that killed the service.

zellneralex, 2021-02-06:
Lucky and at the same time unlucky day:

pi@Voron350:~ $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated)
   Active: active (exited) since Sat 2021-02-06 07:53:04 CET; 8h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 7473 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
  
 pi@Voron350:~ $ sudo systemctl status klipper
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated)
   Active: active (exited) since Sat 2021-02-06 07:53:04 CET; 8h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 7473 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

zellneralex, 2021-02-06:
Mainsail output: Moonraker can't connect to Klippy!
Please check if the Klipper service is running and an UDS (Unix Domain Socket) is configured.

zellneralex, 2021-02-14:
HVe seen the event 2-3 times more. No new reason neither in system ctrl nor in the klippy.log

Arksine, 2021-12-22:
Given that Power devices now support "bound services" I believe that this issue can be closed as resolved.

closed 2021-12-22 by Arksine