Arksine/moonraker #84: Support for Shelly or even MQTT

nicedevil007, 2021-02-06:
Heya, would it be possible to add a Shelly compatibility like you did with Tasmota or TP-Link devices?

Most compatibility would we done with MQTT support :) than we can just add nearly everything to turn on/off the printer 💪

Arksine, 2021-02-06:
While the protocol looks simple enough, I can only add support for devices in which I can test myself.  Support for other devices would need to be contributed (Tasmota was).    I do plan on adding some kind of MQTT support in the future for notifications and there will be hooks for the power plugin.  I don't have a specific timetable on it though.

This overlaps with #30, so I would respectfully ask that future discussion on  MQTT integration move there.

nicedevil007, 2021-02-06:
Mhhh if I find the right place where you did the part with tasmota I should be able to get it done for Shelly. Than you can integrate it... unfortunately I'm not that familiar with your code 😂 will try and test and then report back!

nicedevil007, 2021-02-06:
Ok took a look on your tasmota part but didn't get it.... so we can close this and maybe at any time you will release MQTT integration.

closed 2021-02-06 by nicedevil007