Arksine/moonraker #86: update_manager doesn't load anymore

gdachs, 2021-02-06:
`2021-02-06 14:18:21,685 [moonraker.py:load_plugin()] - Unable to load plugin (update_manager)
Traceback (most recent call last):
  File "/home/gdachs/moonraker/moonraker/moonraker.py", line 148, in load_plugin
    plugin = load_func(config)
  File "/home/gdachs/moonraker/moonraker/plugins/update_manager.py", line 970, in load_plugin
    return UpdateManager(config)
  File "/home/gdachs/moonraker/moonraker/plugins/update_manager.py", line 53, in __init__
    "moonraker": GitUpdater(self, mooncfg, MOONRAKER_PATH, env)
  File "/home/gdachs/moonraker/moonraker/plugins/update_manager.py", line 450, in __init__
    raise config.error("No match for 'env_package_path': %s"
confighelper.ConfigError: No match for 'env_package_path': /home/gdachs/moonraker-env/lib/python*/site-packages`
It is not clear what I can do regarding this. I don't understand what 'env_package_path' is matched against.

Arksine, 2021-02-06:
Please provide the full log.  Thanks.

gdachs, 2021-02-06:
done

Arksine, 2021-02-06:
Thanks.  Can you ssh into the Pi and post the output from the following commands?
```
ls ~/moonraker-env/lib
```
and
```
ls ~/moonraker-env/lib/python*
```
Thanks.  There appears to be something unexpected about your env installation and I need to see if I can accommodate it.

gdachs, 2021-02-06:
Never mind. The problem was the symlink I needed for another issue. I removed it and it works fine, thanks.

closed 2021-02-06 by gdachs

Arksine, 2021-02-06:
Great, thanks.  I think that this occurred as a result of the fix for that issue.

gdachs, 2021-02-06:
Yes, that came to my mind so I thought I give it a try

gdachs, 2021-02-06:
But still what are you matching against? What does the part of code check that raises the exception? I am not so good with python. I am used to C/C++.

gdachs, 2021-02-06:
Maybe it's because you get 2 hits if you check with that pattern the file system?

Arksine, 2021-02-06:
Right, what I am doing is using python's "glob" module to match a pattern, specifically `/home/gdachs/moonraker-env/lib/python*/site-packages`.  This allows Moonraker to successfully detect the "site-packages" directory regardless of the python version.  However,  Moonraker only expects one match, if it gets more than one it doesn't know which one to use.  The symlink was providing Moonraker two matches, which generated the error.