Arksine/moonraker #87: Using api-key

markwinger, 2021-02-06:
I am attempting to setup remote access to fluidd using port forwarding.  When I try to connect, I get the add printer dialog and cannot get any further.  I tried adding $key={api-key} to the url, but that did not seem to do anything.

Since the remote access device I'm using will never have a static ip-address I assume I must somehow present the api-key when connecting.  Am I going about this the right way?  Is this even possible?

Arksine, 2021-02-10:
I don't believe that Fluidd supports using the API Key.  Note that the API Key can not be added to the query string, it has to be added via the `X-API-Key` request header.

Clients will need to add support for the API Key in order to access Moonraker from a non-trusted IP.

Arksine, 2021-04-21:
I'm going to close this as it needs to be implement by clients.  However in the future they will likely authenticate with JSON Web Tokens, which are now supported.

closed 2021-04-21 by Arksine