Arksine/moonraker #88: Update Manager with KlipperScreen seems not to work

nicedevil007, 2021-02-07:
Good morning,

tryed to get KlipperScreen inside your update manager. It tells me invalid configuration. I did everything that @jordanruthe told us in his wiki but still "invalid".

May I reference to this https://github.com/jordanruthe/KlipperScreen/issues/75.

I hope one of you can tell me what is wrong with those lines of code.

Arksine, 2021-02-07:
This suggests that there is something about your repo that could cause issues with an update.  If you could attach moonraker.log it will tell me what is wrong with it and I can provide further guidance.

zellneralex, 2021-02-07:
SSH in your PI and then
cd KlipperScreen
git status
This will show your problem, and has nothing to do with moonraker at all

nicedevil007, 2021-02-07:
here is the output:

```
pi@klipper:~/KlipperScreen $ git status
HEAD detached from ad17131
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   ks_includes/KlipperScreen.conf

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        ks_includes/__pycache__/
        panels/__pycache__/

no changes added to commit (use "git add" and/or "git commit -a")

```

closed 2021-02-07 by nicedevil007

zellneralex, 2021-02-07:
There is your problem. Make a copy of that file place it at ~/ and make your modifications there. KlipperScreen will read that see [https://github.com/jordanruthe/KlipperScreen/blob/master/docs/Configuration.md](url)