Arksine/moonraker #91: Tasmota on Sonoff Basic

masgar, 2021-02-09:
I have a problem with my sonoff basic with tasmota. Power ON/OFF works fine but to get the correct status I had to change the class Tasmota in power.py

from:

self.output_id = config.getint("output_id", 1)

to 

self.output_id = config.getint("output_id", "")

because my sonoff always returns POWER: and never POWER1:

Arksine, 2021-02-10:
Presumably you need to enable `SetOption26` in Tasmota as outlined in [configuration.md](https://github.com/Arksine/moonraker/blob/master/docs/configuration.md#power).

masgar, 2021-02-10:
Shame on me. I'm new with tasmota, used to micropython or espeasy. RTFM is always the best way. With SetOption26 enabled it works flawlessly

Tnanks.

closed 2021-02-10 by masgar