Arksine/moonraker #92: script sudo_fix.sh  Error: visudo: Command not found.

anjoku, 2021-02-09:
Hello,
I'm using an Orange PI LT with: Armbian 21.02.1 Buster with Linux 5.10.12-sunxi

I already had the problem that I couldn't reboot or shutdown from the web interface
https://github.com/Arksine/moonraker/issues/23#issue-714888519
The workaround helped me at the time:
echo 'pi ALL = (ALL) NOPASSWD: / sbin / shutdown, / sbin / reboot' | sudo EDITOR = 'tee -a' visudo

well, about 3 months later I use a fresh installation with KIAUH and Moonraker and fluidd. Without octoprint.

Since a reboot and a shutdown did not work again, I wanted to run the script sudo_fix.sh to fix the problem.

But unfortunately I get the following errors:

andre @ orangepilite: ~ $ ./moonraker/scripts/sudo_fix.sh
###### File not found: 020-sudo-for-moonraker
###### Creating 020-sudo-for-moonraker ...
###### ... done
###### Verifying Syntax of 020-sudo-for-moonraker
./moonraker/scripts/sudo_fix.sh: Line 75: visudo: Command not found.
###### Syntax Error: Check File: / tmp / 020-sudo-for-moonraker

Because I couldn't help myself, I first did that again:

echo 'pi ALL = (ALL) NOPASSWD: / sbin / shutdown, / sbin / reboot' | sudo EDITOR = 'tee -a' visudo

But I have read that this is not so sure.

Best regards

Andre
[moonraker.log](https://github.com/Arksine/moonraker/files/5951277/moonraker.log)

Arksine, 2021-02-10:
Hi anjoku,

Sadly the `sudo_fix.sh` script is intended for CustomPiOS derived distros (such as Octopi).  It isn't surprising that it does not work on Armbian.  I don't own any devices that run Armbian so I can not provide any official support for them.  If someone is able to come up with a script that gives Moonraker permission to run sudo without a password on Armbian I would be happy to review it and merge if it looks good.

anjoku, 2021-02-11:
Hello,

but can I not adjust the sudoers so that the script is only executed without errors for the short time the script is running?

Or what is the script doing? does it only enter a certain line in sudoers?
I could enter them myself with visudo, if I knew something.

thank you andre

Arksine, 2021-05-10:
I'm unsure if this issue got resolved.  I'm going to go ahead and close this as the Orange Pi is not officially supported and it would require an interested developer to add detailed support for Moonraker.

closed 2021-05-10 by Arksine

l3VGV, 2021-10-05:
Im on orangepipc2, user is newly created "klipper". it can sudo, but visudo is not found and script cant run.
Try that, go to ur ssh shell with, login as klipper user, and add to path
PATH=$PATH:/usr/sbin:/sbin

then scripts should work fine. and shutdown woks too!