Arksine/moonraker #93: Shelly Support <3

nicedevil007, 2021-02-09:
Hey Arksine... you are the man!

Thank you very much for the shelly integration.
It is working, just tested it.... just told a friend that you can't release shelly because you don't own them... and was clicking through your repo.... then I found that you added it silently <3

Ok, because you can't test, I will do it for you :)

I guess what is missing is a command to get the shelly's actual state from time to time.

You get it with the IP Adress of the device and `/status` at the end.
Like this: http://192.168.1.75/status

Then a json would be sent back:

```
{
    "wifi_sta": {
        "connected": true,
        "ssid": "Printer",
        "ip": "192.168.1.75",
        "rssi": -62
    },
    "cloud": {
        "enabled": false,
        "connected": false
    },
    "mqtt": {
        "connected": true
    },
    "time": "21:49",
    "unixtime": 1612903777,
    "serial": 13976,
    "has_update": false,
    "mac": "XXXXXXXXXXX",
    "cfg_changed_cnt": 1,
    "actions_stats": {
        "skipped": 0
    },
    "relays": [
        {
            "ison": true,
            "has_timer": false,
            "timer_started": 0,
            "timer_duration": 0,
            "timer_remaining": 0,
            "overpower": false,
            "source": "http"
        }
    ],
    "meters": [
        {
            "power": 0.02,
            "overpower": 0,
            "is_valid": true,
            "timestamp": 1612907377,
            "counters": [
                0,
                0,
                2.589
            ],
            "total": 134009
        }
    ],
    "temperature": 34.21,
    "overtemperature": false,
    "tmp": {
        "tC": 34.21,
        "tF": 93.58,
        "is_valid": true
    },
    "update": {
        "status": "idle",
        "has_update": false,
        "new_version": "20210115-103659/v1.9.4@e2732e05",
        "old_version": "20210115-103659/v1.9.4@e2732e05",
        "beta_version": "20210122-155351/v1.10.0-rc1@00eeaa9b"
    },
    "ram_total": 51040,
    "ram_free": 37848,
    "fs_size": 233681,
    "fs_free": 161393,
    "uptime": 155741
}
```

So you see in "relays" at "ison" you get the acutal state of the switch. Maybe that helps.
Just for future "Enhancements".

Would it also be possible to show the actual power measurements of devices that supports this?
"meter" => "power" is the actual power consumption in "Watt"
"meter" => "total" is the over all power consumption in "Watt per minute" => to get Wh we just have to devide this by 60.

Arksine, 2021-02-10:
Hi nicedevil, Shelly support was contributed by @zellneralex.  Generally speaking, the power module simply exists to toggle the power.  It queries basic state (is it on or off?) so it can be initialized and properly execute some of the automated power events supplied by Moonraker.   Beyond that, I don't want to get into adding specific support for each device as it could complicate the module quickly.

zellneralex, 2021-02-10:
The Shelly support is added in the same way as the support of the other devices. That means you can toggle a relay and get back the state of it (ison is used for that). What you ask would sure be possible, done that in my SmartHome interface, but as @Arksine  said outside of the scope of moonraker.

nicedevil007, 2021-02-10:
@zellneralex Danke :)
Anyways.... good job both of you!

I was asking the "status" question because there are multiple ways to switch the plug on or off.
F.e. by a button next to the printer or by voice command.

Oh.... while writing this now.... maybe you already added an option to send a "push" command to the moonraker api that turns the switch on and off? If this is the case, we just have to "recode" the part from the smarthome stuff to send the command to moonraker instead directly to the shelly :)

zellneralex, 2021-02-10:
Ok, moonraker has already 3 commands:
on -> switch on and read relay state
off -> switch off and read relay state
info -> read relay state

I use MQTT for my SmartHome stuff and can see the changed state there any time I do switch it in the web UI. If I do the switch from "outside" the WEB UI does not show the change. The reason for it, is that neither moonraker polls nor the WEB UI polls info command regularly.
Again moonraker has basic support no fancy stuff like push forwarding.

A solution could be that the Web UI calls "info" when you open the menu but that will costs time and will look like the menu is lagging. 

But honestly I do not see that as a big problem most of the time you will use klipper macros to control that switches anyhow. In klipper you are not able to read that state either. The only time I switch manually is to Power On the 24V PS 

Honestly best solution would be to add a simple MQTT client to switch an relay than you would not need to support all devices at its own and would not need to handle push and pull. MQTT is a standard that nearly all device of this kind support. But that outside of my Python knowledge. I know there is a complete lib for Python (paho-mqtt) and I might do tries with it sometimes.
But also than I would limit it to on/off/RB state

nicedevil007, 2021-02-10:
> Ok, moonraker has already 3 commands:
> on -> switch on and read relay state
> off -> switch off and read relay state
> info -> read relay state
>
> I use MQTT for my SmartHome stuff and can see the changed [...]

The part with MQTT is the same for me, or even ioBroker and Node-RED :)

So what my thought was, just as an example:
If I push a button on my Node-RED Dashboard or send a command via Alexa I send a HTTP or MQTT message directly to the shelly device. But I can change this behaviour that these commands should send a moonraker API call to let this toggle the powerstate of the plug.

Then the WebUI of Fluidd or Mainsail would always be correct and Node-RED Dashboard as well, because they get those changed states automatical from MQTT or ioBroker(HTTP).

zellneralex, 2021-02-10:
As said not supported. Feel free to make changes an talk to @Arksine to may get it in moonraker. I have no plans to to any change

nicedevil007, 2021-02-10:
I believe you didn't understand what I tryed to say xD

The functionality I'm looking for is already there, nothing has to be coded. I just have to change Node-RED commands for my aim :)

Arksine, 2021-02-10:
FWIW,  I do plan look in to support for MQTT notifications and provide hooks into the power plugin for them, however I don't have a timeline on it as its a lower priority than some other things I am working on.

closed 2021-07-03 by nicedevil007