Arksine/moonraker #95: Feature request: PanelDue support for v3.2+ firmware

Norbs12, 2021-02-11:
The newer firmware has some useful features.

One that really stands out is you can now control one other heater which is useful for monitoring chamber or maybe a second extruder.

BastianGschrey, 2021-02-26:
+1
3.2 would be really helpfull.

Arksine, 2021-02-26:
I plan to look at this at some point, but it isn't a high priority.  Version 3.2 represents a significant change in the API therefore the plugin will need a lot of attention to support it, and it needs to be done without breaking compatibility with version 1.24.

trevjonez, 2021-03-01:
>  it needs to be done without breaking compatibility with version 1.24.

do we have data on those with panels that can't support 3.2 and are using moonraker+klipper? seems like if the croud size is small enough it would be best to just EOL support to ease the workload

Aianir, 2021-05-23:
Please keep supporting the PanelDue. I have one printer with PanelDue and one with the stock Ender 3 display. The PanelDue is far better when it comes to actually doing something on the printer, not just displaying progress.

Arksine, 2021-05-23:
I don't have any intention of dropping support for the PanelDue.  I will add support for version 3.2 when I have the opportunity, however as previously mentioned it isn't a priority and not something I have near term plans of working on.

SirDeath81, 2026-08-17:
Support for Paneldue Firmware 3.7.0 is on its way. Programming is completed, first tests with 3.7.0 all ok.
complete tests with Paneldue with Firmware 3.7.0 and 1.25.0 to follow within this week. If I'm really successful, I'll do a fork and then a Pull Request.
The only thing I want to ask you is a little bit of patience with me - this is my first time ever contributing something on github :-)

EDIT: The new Version of paneldue.py will support both Firmware Versions with autodetect of the connected Version so we don't destroy any existing installations :-)