Arksine/moonraker #96: Verify G-Code files after upload

TimoErdmann, 2021-02-14:
My internal network seems unstable sometimes and some of my gcode files get corrupted while uploading it to my printers. Most of my prints are rather long and need over 24 hours to print. 

It would be nice if the files would be checked after the upload. So cancelled prints can be prevented.

At the moment the print is started nevertheless the file is corrupted. When the corrupted line is reached the printer stops and needs to be restarted with a freshly uploaded file which hopefully has no defects.
[image]

I dont know if this this is the correct place or mainsail, fluidd or even klipper should be responsible for the verification of the files. So please have mercy if this is not the right place.

Arksine, 2021-02-26:
Data integrity of the upload shouldn't be an issue, if the request is dropped due to an unstable connection then it should error out and never write the file to disk.

The issue in your example appears to be a problem with the slicer.  It seems to have inserted a "," in place of "." to represent the decimal for the E axis position.

TimoErdmann, 2021-02-27:
Yes in the most cases you may be right but the connection seems to be pretty unrielable. I have a dlan system and it seems that there are sometimes bitshifts which are not noticed. 
`
G1 X131.723 Y95.161 E3501.25736
G1 X131.71 Y94.81 E3501.27543
1 X031.696 Y94.46!E3501.69345
G1 X131.68 Y)4.111 E3501.31143
G1 X131.667 Y93.779$E3501.329u5
G1 X131.658 Y93.404 E3501.34782
G1 X131.657 Y93.294 E3501.35348
G0 F3360 X131.u5 Y;3.043
G0 X131.57 Y92.834
G0 F21000 X131.698 ٸ8.668
G1 f4800 X139.696`Y88>628`E3509.3554
+G1 X121.78(Y88.279 E3501.37351
Ǳ X131.667 Y97.927 E3501.39164`

This is a part of the GCode file after the upload to Klipper. You can see it is not only the conversion from the dot to a comma. I would exclude the slicer as a problem. Every other gcode I create is working just fine and I generate many files over the day to serve at least 8 printers a day.  

So why don't we add a verification of the uploaded GCode to prevent a broken print after many hours of printing time. Otherwise it will result in much waste of plastic and lots of lost hours of lifetime. I know it won't happen often and it really rare. 
Maybe I can add this feature by myself, but I will be able to do it in 2 month at the earliest.

Arksine, 2021-04-21:
It is now possible for clients to include an SHA256 checksum with the upload.  Clients will need to implement this, however it is functional in Moonraker.

closed 2021-04-21 by Arksine

beebenutz, 2024-04-21:
> It is now possible for clients to include an SHA256 checksum with the upload. Clients will need to implement this, however it is functional in Moonraker.

Can you please elaborate on this? I have some small issues which I believe is caused by file integrity. Implementing the SHA256 checksum would likely remedy my issues. I looked around the web, but couldn't fine any literature on the subject.

Thanks, Mat

Arksine, 2024-04-21:
Frontends can add a SHA256 checksum to the form.  For example, the payload of the upload request might look like the following (sans the binary data):
```
------WebKitFormBoundaryR0GCxagNvZNE8YFm
Content-Disposition: form-data; name="file"; filename="Support_test.gcode"
Content-Type: text/x.gcode

------WebKitFormBoundaryR0GCxagNvZNE8YFm
Content-Disposition: form-data; name="root"

gcodes
------WebKitFormBoundaryR0GCxagNvZNE8YFm
Content-Disposition: form-data; name="checksum"

a24193ace930f2d6d985f1e8648cb20bd3116d221906721dc0e6db66acabddc5
------WebKitFormBoundaryR0GCxagNvZNE8YFm--
```

The programming language/frameworks should provide an abstraction for adding items to the form.  The moontest repo [shows how to do it](https://github.com/Arksine/moontest/blob/master/src/js/main.js#L2201) using a JS FormData object.