Arksine/moonraker #98: mcu_host doesn't survive a printer power cycle

Loreagle, 2021-02-19:
Hello, I'm writing here since I can't find a solution to this problem and I think it could be moonraker related.
I have an Orange Pi PC Plus connected to my MKS Robin Nano and everything works, I also configured power plugin in moonraker using MKS relay module and it works correctly.

The problem started when I configured the virtual mcu for using the accelerometer in Klipper, if I enable [mcu rpi] in my printer config it works perfectly at first power on of the printer but, if I switch printer off and then on again the virtual klipper process will be killed and I have the error "Can not update MCU 'rpi' config as it is shutdown".

If I open the terminal and digit:

`sudo service klipper stop`
`sudo service klipper start`

then it works.

I also tried different settings in power plugin in moonraker like estart_klipper_when_powered on true or false but it seems to me that only restart the normal klipper and not the virtual one.

Attached here my logs:
[moonraker.txt](https://github.com/Arksine/moonraker/files/6009326/moonraker.txt)
[klippy.txt](https://github.com/Arksine/moonraker/files/6009327/klippy.txt)

Loreagle, 2021-02-19:
[image]

Arksine, 2021-02-19:
Moonraker is not killing the host process.  As can be seen from your output, Klippy received a HUP signal that killed it.  I suspect that this is the same thing that is occurring in #83.

Loreagle, 2021-02-19:
> Moonraker is not killing the host process. As can be seen from your output, Klippy received a HUP signal that killed it. I suspect that this is the same thing that is occurring in #83.

To fix it is not possible to run a `sudo service klipper start` when restart_klipper_when_powered is on True?

Loreagle, 2021-03-01:
@Arksine After latest Fluidd update (v1.9.0) I can now correctly communicate with the printer pushing "RESTART KLIPPER SERVICE" button with printer powered ON. The problem is that the option "restart_klipper_when_powered: True" doesn't work at the same way. Is there a way to fix it?

Arksine, 2021-03-01:
The "restart when powered" option does not restart the Klipper service, it issues "firmware_restart".  Issuing a service restart here would be problematic in scenarios where it isn't necessary. 

It is possible that this issue will be resolved in Klipper, so the service does not hang when you issue a restart.  See KevinOConnor/klipper#3996.

Arksine, 2021-04-10:
Presumably this should be solved in Klipper, so I'm going to close this.

closed 2021-04-10 by Arksine